steveicarus/iverilog #53: {256'b0}[ADDR-1:0]; Syntax error

abhisheietk, 2015-02-18:
I am not sure how to define in.

But in order to avoid ISE warning I am using this construct. It works their but gives error in  iverilog.

Address_In	<=	{Address_In  + 1}[ADDR-1:0];

../../hdl/fifo.v:73: syntax error
../../hdl/fifo.v:73: error: malformed statement

and

{256'b0}[ADDR-1:0];

../../hdl/fifo.v:47: syntax error
../../hdl/fifo.v:47: error: invalid port connection expression.

martinwhitaker, 2015-02-18:
Taking a subrange of a concatenation is a SystemVerilog feature, and is not currently supported by Icarus (support for SystemVerilog is work in progress).

Does ISE give a warning if you write

    Address_In <= Address_In + 1'd1;

(it shouldn't, as the size of the LHS and RHS of the assignment are the same in this case)

abhisheietk, 2015-02-18:
Thanks Martin

Address_In <= Address_In + 1'd1; worked...........

 but how to go about 

.b_port_data_IN({256'b0}[DATA-1:0]),

in

dpram	Fifomem	(
	.clK(clK),	

	.a_port_WR(FIFO_wr),
	.a_port_ADDR(Address_In),
	.a_port_data_IN(fifo_IN),
	.a_port_data_OUT(),

	.b_port_WR(1'b0),
	.b_port_ADDR(Address_Out),
	`ifndef SIMULATION
	.b_port_data_IN({256'b0}[DATA-1:0]),
	`else
	.b_port_data_IN(),
	`endif
	.b_port_data_OUT(fifo_OUT)
	);
	defparam	Fifomem.DATA	=	DATA;
	defparam	Fifomem.ADDR	=	ADDR;

abhisheietk, 2015-02-18:
.b_port_data_IN({256'b0}[DATA-1:0]),

.b_port_data_IN(), I am ok with this also but wanted to avoid a warning from ISE.

any suggestions

martinwhitaker, 2015-02-18:
Either

    localparam [DATA-1:0] ZeroData = 0;
    .b_port_data_IN(ZeroData),

or

    .b_port_data_IN({DATA{1'b0}}),

abhisheietk, 2015-02-19:
Thanks Martin !!

closed 2015-02-19 by abhisheietk