steveicarus/iverilog #58: unability to async-reset wide registers to non-zero value

klammerj, 2015-04-02:
I run the command:
``iverilog -tblif ./dff_scramble.v``

The error message I get is:
``./dff_scramble.v:24: sorry: Forgot how to implement asynchronous scramble (set to x/z).
./dff_scramble.v:23: error: Unable to synthesize synchronous process.
``
It happens in synth2.cc

The file is:
```verilog

/*
 * A D-type flip-flop to check synchronous logic works
 * correctly.
 */

module dff(
output wire [3:0] q, 
input wire [3:0] d, 
input wire clk, 
input wire rstin,
input wire ce,
input wire [3:0] oe);

reg [3:0] rg;

bufif1(q[0],rg[0],oe[0]);
bufif1(q[1],rg[1],oe[1]);
bufif1(q[2],rg[2],oe[2]);
bufif1(q[3],rg[3],oe[3]);

  always @(posedge clk or posedge rstin)
    if (rstin)
      rg <= 4'b0001;
    else
	    if(ce)
  	    rg <= d;
  

endmodule // dff

```

steveicarus, 2015-04-02:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

The -tblif target is a synthesis target, so it uses the synthesizer
to get down to blif output. (I assume you know what that is, and
that is why you are using -tblif in the first place.)

It appears that the synthesizer cannot handle reset values that
are not all ones or all zeros. That is because it is trying to
generate a DFF with asynchronous set/reset instead of asynchronous
load. That is in theory possible to fix, but...

You will find that the blif target does not support DFF with
asynchronous set/reset/load. When implementing the blif target,
I found that the ABC tool chain (the intended user of the blif
output) did not really seem to support DFF devices at all. I tried
to get better information from the ABC developers, but couldn't
get a response, and my customer at the time came to realize that
given its limitations, ABC wasn't going to suit their needs.

So what's your interest in BLIF? Is this worth my digging into
in more detail?

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlUdY1sACgkQrPt1Sc2b3ilNNgCcCp10X3swuVTP7iHte/JjH6e3
iN8AoON/3fJ8MmDCBCgpkT/QxFsesOnP
=CXiW
-----END PGP SIGNATURE-----

klammerj, 2015-04-02:
The -tblif was just for illustrative purposes. I was actually using -tatf a target for atmels PLDs. 
You can find it in my iverilog <https://github.com/klammerj/iverilog> fork. Coincidentally, I was also using ABC, albeit for combinatorial optimization only. The target uses the FF/sync-synthesis, and seems functional (for the very few things I've tried). 
The broken FFs are one of the few sore spots left. 
Feel free to merge... It's a bit outdated tho, and I've tampered with other things, too...

klammerj, 2015-04-02:
PS: I just looked into the abc directory here. It seems they support the .latch primitive(it's also for edge sens. stuff)

klammerj, 2015-04-09:
I've whipped up a patch that fixes this one and #34
http://members.aon.at/aklamme4/scratch/0001-lhs-partsel-and-sync-scramble.patch
(why does github mangle the '~' character so the link won't work when it's in there?)
http://members.aon.at/~aklamme4/scratch/0001-lhs-partsel-and-sync-scramble.patch

caryr, 2015-04-09:
No Idea on the '~' character. I thought you could also add patches to these reports.

Thanks for the patch. I will look at your patch and if I don't see any problems I will apply and push it to master. Hopefully I can find some time over the next day or so to get this done.

caryr, 2015-04-10:
I started looking at your patch and while trying to verify #34 I ran into a problem in that it reports that an Asynchronous LOAD is not implemented. Have you verified that the code you posted for #34 works for you? I didn't poke around too much, but it looks like the constant value is being split with a VP for each bit which is then not being reported as driven by the driven_mask() function.

klammerj, 2015-04-10:
Weird. The code posted  #34 segfaults on my branch now... maybe the missing clk...
a different one which uses clk, works on my local branch but fails on upstream+patch. 
I'll try to find out If I locally changed any other related files besides synth2.cc. be patient.

klammerj, 2015-04-10:
It's this one... 
http://members.aon.at/aklamme4/scratch/0002-driven_mask.patch

caryr, 2015-04-10:
Okay I will check that patch out as well.

As a side comment; it looks like you are not following our indenting/coding style. It would be best if you could do that for future patches. I will submit a follow up patch once I have confirmed your patches are working correctly to fix that. Once again thank you for the patches.

caryr, 2015-04-10:
After applying the patch the warning goes away, but I get a segmentation fault (I'm just using -S though -t blif had the same issue). I will look into this when I have some free time to compile a debug version of the code. Hopefully over the weekend. Using valgrind with my existing executable points to an uninitialized value being used in a condition in Nexus::driven_vector() and the segmentation fault is in NetCondit::synth_sync(). Though with optimization and no debug information you never know for sure.

klammerj, 2015-04-11:
The segfaults I've been able to observe were in: 
synth2.cc around line 1602 if there's no else to an if. 
and around line 1816 (events vector was NULL.) which may have been triggered by some oops in sensitivity list... 
haven't run valgrind on it recently...
wrt coding style, I've not been able to find documentation or an indent line.. (SF seems down now, so can't search the mailarchive).

klammerj, 2015-04-11:
It's back...
I've found this...
http://sourceforge.net/p/iverilog/patches/139/#4edb

another patch:
http://members.aon.at/aklamme4/scratch/0003-indent-i6-ts8-lp-bli0.patch

(Not sure you really want that, tho...)

klammerj, 2015-04-11:
don't apply that last one, It seems indent puked in there.

klammerj, 2015-04-11:
manually edited and regenerated the first two patches...
from what I can tell, the indentation is OK now for 0001 and 0002...

caryr, 2015-04-20:
I have applied the two patches and did some minor cleanup, but I think there needs to be some more changes. I will attach the example that I am using to test #34 and since it is just resetting the D-FF it should not generate both a set and clear connection. Even the attached example should generate three FFs with a clear and one with a set. If the code could do that then we could actually simulate the synthesized code using vvp. It would also reduce the amount of dead code generated to drive the unneeded input to a low level.

I also added a more definitive warning that vvp does not currently support a D-FF with both an asynchronous set and clear. I may be able to add this, but at the moment I wanted to get these patches pushed so you could look at the changes I am suggesting.

caryr, 2015-04-20:
Here is the code I was testing I was using iverilog -S "filename". I'm assuming this should just generate three D-FFs that have an asynchronous clear. If you look at the a.out output you will notice a significant amount of extra stuff and that is what I was hoping could be eliminated. I'm guessing you may also have some of the same extra stuff in your output code.

```verilog
module pci_tgt(input wire [2:0] d,
               input wire clk,
               input wire RST,
               output reg S0,
               output reg S1,
               output reg S2);

  `define S {S0,S1,S2}

  always @(posedge clk or posedge RST) begin
    if (RST) `S <= 0;
    else `S <= d;
  end

endmodule

module test;
  reg [2:0] d;
  reg clk, rst;
  wire S0, S1, S2;

  pci_tgt dut(d, clk, rst, S0, S1, S2);

  (* ivl_synthesis_off *)
  initial begin
    $monitor(S0,S1,S2,,d,,clk,,rst);
    clk = 1'b0;
    rst = 1'b0;
    #1;
    rst = 1'b1;
    #1;
    rst = 1'b0;
    #1;
  end
endmodule
```

klammerj, 2015-04-21:
yes, it concats even single bits....should do sthg about it...
not sure why it happens, tho... it's single bit regs....
Another thing I just ran into: when I get all zeroes or all ones, the reset input 
is just one bit wide, which confused the fitter here. Should there be a convention 
regarding the width of those inputs for wide registers? also clock and clken(I'm 
currently using single bit ones in the back-end even for wide regs)
(removing the all-ones and all-zeroes clauses makes the reset/set inputs wide in all cases)

klammerj, 2015-04-21:
I've been doing something wrong...

>(gdb) p *rst_drv.bits_
>$7 = verinum::Vz

klammerj, 2015-04-21:
Nope, wasn't me...
there's a 3'sd0 const _s5 that outputs a sig _s4
which goes in a VP _s7, 1 bit comes out as _s6....
driven_value reports it as hi-Z.. it should not...
would have to walk across those PV things...

klammerj, 2015-04-21:
Just pushed a commit on my branch. It works with your test, but is still somewhat broken 
WRT TRI0/1 nets....
klammerj/iverilog@e1d7712aed97de9c2cb18916439c352258310471

caryr, 2015-04-22:
Okay, It's good you are making progress. Make sure you sync to master since I made some formatting and space fixes in the code you originally submitted.

klammerj, 2015-04-23:
http://members.aon.at/aklamme4/scratch/ff_patch2.tbz
These patches are against your current HEAD(I hope).
They fix the broken driven_vector function, and produce valid 
aset_value members in the NetFFs. The regression tests complete 
with 5 fails again as before the patches... tgt-vlog95 expects 
single-bit set/resets and aset_value.

caryr, 2015-04-24:
Okay, I will look at these patches.

I will also see if I can fix the vlog95 issue. The correct way is probably to create an arrayed UDP instance, but the Icarus front end does not currently support this so it cannot be tested full circle.

klammerj, 2015-05-21:
Do you need the patches refreshed, or a pull request or something?

martinwhitaker, 2015-06-05:
Sorry we haven't responded to this sooner. I'm reviewing the patches now.

martinwhitaker, 2015-06-08:
OK, I've reviewed your patches. I wasn't happy with the first two patches for several reasons:
* I'd expect drivers_constant() to return true if driven_vector() returned a constant value.
* You removed some of the checks I put in place to protect against multiple drivers. We could handle these, but to be absolutely safe, we would have to check and resolve drive strengths.
* I'm a bit reluctant to make changes that might introduce subtle bugs in constant propagation this close to the v10 release.

Instead I've reworked the concatenated lval synthesis. It now splits up the concatenated parts and performs constant propagation prior to performing the synthesis. This gives the output you would hope for for Cary's example without any further work.

I've applied your third and fourth patch (and then done a bit of simplification) as I think that is the right approach. Note that I only create an aset_value when it is needed (the LPM spec. says it defaults to all 1's if not specified). You can check for the default case by checking if the verinum length is zero.

There's still quite a bit of work to do:
* tgt-vvp and vvp need to be modified to handle the aset_value.
* the behaviour of the dff functor in vvp needs to be fixed (currently it performs an asynchronous set/clear on either edge of the rst signal, and ignores the state of rst in the meantime).
* the driven_mask() function is used to check for non-constant reset values, but the implementation of driven_mask() actually determines whether the bits are driven at all, not whether their values are constant. I think we should be using drivers_constant() here.
* It's still far too easy to get a segmentation fault if you write code in a form the compiler doesn't expect.

I've pushed the changes so far, as they don't break anything that wasn't already broken. I'll try to fix the vvp behaviour in the next few days.

klammerj, 2015-06-11:
thanks, seems to work now...

closed 2015-06-11 by klammerj

martinwhitaker, 2015-06-13:
I've fixed vvp to support aset values and to correctly model asynchronous set/reset behaviour. I've also updated synthesis to support negative-edge triggered flip-flops and to correctly detect and report non-constant asynchronous set/reset values.

I've also changed the way the ivl_target API reports default aset values. ivl_lpm_aset_value will now return null for the default (all 1's) value, rather than returning an ivl_expr_t value with a width of zero. This is what the vlog95 backend was expecting, and I think it is better behaviour.