steveicarus/iverilog #59: VPI: update

themperek, 2015-04-10:
Disclaimer: I do not know internals of simulator or vpi. Possibly doing something wrong.

Test files can be found here:
https://gist.github.com/themperek/4f0053269c7c7838041f

Output:
```
B0 = zz B1 = 01 @ 0
B0 = zz B1 = zz @ 1000
B0 = zz B1 = 01 @ 2000
```
In first order B1 and B0 should be same. There some difference how assign is handled.

The original problem comes from using cocotb. 
The workaround (see comments):
https://github.com/SiLab-Bonn/basil/blob/master/device/modules/utils/bus_to_ip.v#L37
https://github.com/SiLab-Bonn/basil/blob/master/host/basil/utils/sim/SiLibUsbBusDriver.py#L146

PS. The unit test using icarus run pretty nice on travis: https://travis-ci.org/SiLab-Bonn/basil

caryr, 2015-04-10:
I have downloaded your test files and I will try to look at this over the weekend.

From some of the other information you posted it looks like you are using the latest development code so that will be my assumption unless you state otherwise in a follow up message.

caryr, 2015-04-21:
It took me longer than I expected to look at this. The problem is that the compiler is incorrectly connecting the WR and IN terms of the mux created by the continuous assignment to B0 to constant undefined values since that is what it thinks WR and IN are driven by. It does not understand that these can be set by the VPI. It also looks like the assignment to the IN port (at least from the mux perspective) is not working correctly since the value is set correctly and then transitions back to High-Z when I hand edit the compiler output to connect the mux to the IN and WR nets. The procedural code works because it uses an actual reference to the WR and IN handle.

martinwhitaker, 2015-06-22:
I've had a look at this, and the bug is not in the compiler, it's in vvp. The signal_put_value function in vpi_signal.cc was not updated for the changes introduced by the vvp net rework. Rather than sending the value to the net (which sends it to the functor that drives that net), it needs to send it from the net.

I have a fix for this, but need to make sure the same applies to registers, and check the force/release behaviour.

martinwhitaker, 2015-06-26:
Found several more bugs whilst testing this, but it should be working now.

Thanks for reporting this.

closed 2015-06-26 by martinwhitaker