steveicarus/iverilog #60: yosys test repwhile causes crash

frznchckn, 2015-04-13:
yosys commit cliffordwolf/yosys@706631225e9ef0d2954c4bef51aaa2817e0f5e86 contains a test "repwhile" that will crash iverilog

1. run "make" and "make test" to produce all necessary yosys files (the "make test" should fail out at the repwhile test)

2. cd \<yosys_repo\>/tests/simple/repwhile.out

3.  iverilog -s testbench -o repwhile_tb_syn0 repwhile_tb.v repwhile_syn1.v ../../../techlibs/common/simlib.v ../../../techlibs/common/simcells.v 

4. output is as follows
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
ivl: t-dll.cc:542: void dll_target::make_scope_parameters(ivl_scope_t, const NetScope*): Assertion `cur_par->msb >= 0' failed.
Aborted (core dumped)

@cliffordwolf updated yosys with the following commits to avoid the crash
cliffordwolf/yosys@a1c62b79d5d554be86b4b9bd53d72704b045acde
cliffordwolf/yosys@95944eb69e45837516ff9c0cba54f77ab89af754

@cliffordwolf mentioned that the generated code contained questionable constructs, but I thought you would want to know it crashed iverilog.

cliffordwolf, 2015-04-13:
Actually there have been two problems. Both of them only reproducible on some systems. E.g. I cannot reproduce them on 64 Bit Ubuntu 14.04 or 15.04 Beta 2, but I can reproduce them on 32 Bit Ubuntu 14.04. Both of them illegal verilog code that crashes iverilog.

The first issue is related to zero-length parameters:

    module mem;
      parameter WR_CLK_POLARITY = 1'b1;
    endmodule
    
    module test;
      mem  #(.WR_CLK_POLARITY(0'b0)) _0_ ();
    endmodule

On 32 Bit Ubuntu 14.04 and iverilog ba31252 this gives:

    $ iverilog test1.v
    ivl: t-dll.cc:542: void dll_target::make_scope_parameters(ivl_scope_t, const NetScope*):
    Assertion `cur_par->msb >= 0' failed.
    Aborted (core dumped)

According to A.8.7 of IEEE Std 1800-2012 the size part of a number literal must be nonzero. So this is not legal code. But it should print a proper error message, not assert.

The second issue is related to the signedness of parameters being determined by the assigned parameter value. The way I read IEEE Std 1800-2012 sec. 23.10 icarus verilog does the right thing here, but it still could handle cases like the following better but producing a good warning message before running out of memory:

    module mem (WR_EN);
      parameter WIDTH = 32;
      parameter WR_PORTS = 0;
      input [WR_PORTS*WIDTH-1:0] WR_EN;
    endmodule
    
    module test;
      mem  #(.WIDTH(32'd 32)) _0_ ();
    endmodule

Which produces:

    $ iverilog test2.v 
    warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
    draw_net_input.c:711: Error: malloc() ran out of memory.

When "mem" is instantiated as `.WIDTH(32'sd 32)` then the array correctly goes from -1:0 instead of being enormously large. The verinum::as_long() warning is already helpful, but maybe it would be even better to print a proper warning (with a <file>:<line> pointer to the RTL) when a vector size or array dimension > 2**30 is detected.

caryr, 2015-04-14:
Thanks for the clarification Clifford. I was hoping to get a simple Verilog example since tracking bugs/problems using other software is challenging on a good day. The first issue should be fairly trivial to catch and hopefully we can find something sane for the later case.

martinwhitaker, 2015-04-26:
I've pushed fixes for both these issues. A zero size for a sized constant is now treated as an error. I've added a warning for excessive vector and array dimensions, but can't test whether it fixes your particular example, as I don't have a 32-bit system.

caryr, 2015-04-27:
I can confirm on a 32-bit system the following message:

gh60.v:4: warning: Vector size is greater than 1073741824.

 is now generated between the verinum::as_long() warning and the malloc() failure.

Since it appears all the issues have been addressed I will close this report.

closed 2015-04-27 by caryr