steveicarus/iverilog #64: Pause Resume doesn't work twice in a row because the handler gets removed.

knightpraetor, 2015-04-29:
I believe that the OS seems to remove the handler after signalling SIGINT. I have looked through the code and don't see anywhere it is being removed within the Iverilog code, yet I have found several websites online that suggest that you should reattach the handler when the handler is called.

http://www.geeksforgeeks.org/write-a-c-program-that-doesnt-terminate-when-ctrlc-is-pressed/

I'm not sure whether you wish for vvp to support pause and resume or not, but it seems that using the change proposed there that I can have this functionality.

I wanted to be able to pause and resume vvp.exe using ctrl c and the "cont" command, so I built an older snapshot of icarus to try testing it.  

After altering the function "signals_handler" as shown below, I was able to get the functionality I wanted.

schedule.cc

 * These are the signal handling infrastructure. The SIGINT signal
 * leads to an implicit $stop.
 */
extern "C" void signals_handler(int)
{
      //reset signals handler so that program can be paused and resumed with ctrl + c and cont commands
      signal(SIGINT, &signals_handler);
      schedule_stopped_flag = true;
}

Now I am able to hit Ctrl C, pause the simulation, and then send "cont" to standard in and have vvp resume the simulation multiple times.  

If there is a better way of doing this, feel free to let me know

caryr, 2015-04-29:
Which OS are you using? I believe what we have in the existing code does work for the systems we usually use (i.e. ^c and then cont work correctly), though if there are no issues with existing systems I see no reason to not make the change you are suggesting.

knightpraetor, 2015-04-30:
windows 7. when i start a command prompt, run vvp, and then hit ctrl c, it
will pause once, and then i can resume. But pausing it a second time just
has the program exit.

I wasn't sure if a chance was necessary on my system or if i was doing
something wrong, so I just thought I would mention it and see what you
thought.

Thanks.

Josh

knightpraetor, 2015-04-30:
oh one random aside. I wasn't able to build when i cloned the github
because of the following error, so I've been using the 20150105 development
snapshot to try out that change.  I guess it would be better if I could fix
this error, but I'm not really sure how to fix it.

$ g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow
-DSIZEOF_UNSIGNED_LONG=4 -g -MD -c main.cc -o main.o
In file included from pform_types.h:25:0,
                 from netlist.h:38,
                 from pform.h:22,
                 from main.cc:56:
verinum.h:157:8: error: reference to 'ostream' is ambiguous
 extern ostream& operator<< (ostream&, const verinum&);
        ^
verinum.h:28:7: note: candidates are: class ostream
 class ostream;
       ^
In file included from d:\mingw\lib\gcc\mingw32\4.8.1\include\c++\ios:38:0,
                 from d:\mingw\lib\gcc\mingw32\4.8.1\include\c++\ostream:38,
                 from
d:\mingw\lib\gcc\mingw32\4.8.1\include\c++\iostream:39,
                 from main.cc:42:
d:\mingw\lib\gcc\mingw32\4.8.1\include\c++\iosfwd:136:33: note:
    typedef class std::basic_ostream<char> std::ostream
   typedef basic_ostream<char>   ostream;

caryr, 2015-04-30:
Okay, we will try to figure this out.

knightpraetor, 2015-04-30:
thanks. I think I am giving up for now. I have tried 3 different builds but
have yet to get a working version with the change i wanted to allow resume
that will work on windows.

A long time ago I built iverilog 0.9.7 successfully on an XP machine using
mingw/cygwin(can't remember) and was using that successfully.

Now I'm buliding under windows 7 using mingw.

However, the current trunk of github gives me the ostream reference is
ambiguous errors, and I'm not sure how I can fix them. It seems like I can
go through all the files it is complaining about and type std::ostream
instead just ostream and it will remove the error. However, I'm guessing
the more proper way would be to remove the #includes that are resulting in
the ambiguous reference.

I tried building 0.9.7 from the development snapshop since that had worked
in the past for me, but this time I get this error:

lexor_keyword.o:lexor_keyword.cc:(.text+0x14): undefined reference to
`check_identifier(char const*, unsigned int)'
collect2.exe: error: ld returned 1 exit status

Meanwhile I found that I could build the 20150105 development snapshot, so
i built that. It builds fine without errors. However, while iverilog seems
to work correctly. VVP crashes when the simulator attempts to clean up and
shut down.  When i hooked gdb onto vvp.exe, I found that it was failing
inside of ivl_dlfcn.h inside the function:
inline void ivl_dlclose(ivl_dll_t dll)
{
    (void)FreeLibrary((HINSTANCE)dll);
}

It crashes when it attempts to free a handle as the program is closing.

Using the following, I found that the file it was having problems with was
"(icarus_folder)/ivl/system.vpi.
    // HMODULE hmod = (HMODULE)(dll);
    // TCHAR szPath[MAX_PATH - 1];
    // DWORD dwLen = GetModuleFileName(hmod, szPath, MAX_PATH);

I will probably try to spend a bit more time on it because i would really
like to be able to pause and resume icarus so that I can grab simulation
data and not necessarily take up processing time.  However, I don't know
what I should try to get working.

originally I thought I should be using the most recent version to build off
of. However, the old icarus 0.9.7 files that I have on my computer do not
crash from unloading system.vpi, so perhaps I should just try to get that
working.  I feel so lost with that "check_identifier" error though because
after grepping check identifier, I see that the makefile actually creates
the CC file using gperf.
​
So in order to figure out that error I guess I would have to learn how
gperf works.  Seems like it will take a lot of time.  Though writing my own
simulator would take a heck of a lot more time.

Josh

caryr, 2015-04-30:
The latest development tries to clean things up better so that could be related to why the crash is happening. I'm not sure why you are having all the problems with mingw. I regularly compile using cygwin and it supports multiple ^c and cont operations. I will note that in the past I have noticed that a cygwin compiled executable was significantly slower than a mingw executable. The recent snapshots are for development not V0.9. For snapshots we run gperf before generating the tar file so you should not have to know anything about or run gperf. That is not true when you are tracking the latest code from git. Given all that this looks like a compile/link problem not a gperf problem.

I do not have a mingw environment set up at the moment and don't have much free time. I will see if someone else has some time.

knightpraetor, 2015-05-01:
ok, well i will try using cygwin then. Sorry for all the hassle

knightpraetor, 2015-05-01:
cygwin built very quickly and easily off the 0.9.7. I guess that is what I
used before. Thanks

On Fri, May 1, 2015 at 11:25 AM, Joshua Street <knightpraetor@gmail.com>
wrote:

> ok, well i will try using cygwin then. Sorry for all the hassle
>

caryr, 2015-05-01:
We will still try to figure out what is going on with all this since we plan to release the development branch as stable fairly soon. I am glad to hear you have something that is working now.

knightpraetor, 2015-05-01:
hmm, well it's only partially working. However, I think i have a better
guess as to what's going on now.

I know that the configure script doesn't run properly on windows 7, so I
tried to hack it together. However, I am betting that my hack was
insufficient. It's possible that the reason my current build of 0.9.7 has a
segfault on program exit while it doesn't in the old build I did on XP is
because the program is being built incorrectly due to define statements for
the build system not being set correctly.

I was stupidly assuming that since setting the define statement
-D SIZEOF_UNSIGNED_LONG=4  was letting the system build, that it would
build correctly. But looking at it now with a clear head, I doubt that's
the case.

I think if i want a build that will fully work i will need to get the
configure script working or build on an xp machine( i have one i can remote
desktop into, though I may play with it a little first).

One thing that probably should be changed in the configure script is that
it doesn't fail configure and stop when it encounters an error that will
break compilation.
In particular, if size of unsigned long or size of unsigned long long is
equal to zero, I think the configure script should just exit with an error.

Also, in cygwin the error gets disguised, but in mingw i can see where it
fails in the configure script.

cygwin
checking size of unsigned long... 0
checking size of unsigned... 0
checking for tputs in -ltermcap... no
checking for readline in -lreadline... no
checking for add_history in -lhistory... no

mingw
checking size of unsigned long... rm: cannot lstat `conftest.exe':
Permission denied
0
checking size of unsigned... rm: cannot lstat `conftest.exe': Permission
denied
0
checking for tputs in -ltermcap... rm: cannot lstat `conftest.exe':
Permission denied
rm: cannot lstat `conftest.exe': Permission denied
no
checking for readline in -lreadline... rm: cannot lstat `conftest.exe':
Permission denied
rm: cannot lstat `conftest.exe': Permission denied
no
checking for add_history in -lhistory... rm: cannot lstat `conftest.exe':
Permission denied
rm: cannot lstat `conftest.exe': Permission denied
no

Anyway on .97 I got it to build without fixing this by running:
./configure
make

Fails.

don't clean
set CFLAGS and CXXFLAGS with -D SIZEOF_UNSIGNED_LONG=4
configure again.

Make.

this succeeds.

however, just setting CFLAGS and CXX FLAGS at first and then doing
configure and make fails with a different error about gperf.

Basically, I didn't realize that configuring again and then running make
doesn't clean because I'm a freaking noob.

Anyway, I think even with this hack that was letting me build it won't
build a fully working version without those other parameters being set. My
google searches indicated that the conftest.exe issue has something to do
with file handles not being released immediately. So perhaps if I run the
same test outside the configure script and then set the rest of the values
for the tests that fail, then I will be able to build icarus successfully
in windows 7 x64.  I will try that. If not, I will try running it in xp
again because
I definitely didn't have this many problems before when i built it.

Josh

caryr, 2015-05-01:
Configure should work without problems. If you are getting a size of zero for unsigned, etc. then you likely have a problem with your cygwin setup or something cygwin specific. This may be better asked on the cygwin mailing list. FYI cygwin has gperf so is it installed? Did you look at the list of packages needed to compile Icarus on the wiki? cygwin should have everything you need. I will note that I am only testing Icarus on XP 32-bit with cygwin, but I know others have used 7 64-bit so I'm at a loss as to why you are having a problem compiling with cygwin.

knightpraetor, 2015-05-01:
yeah, I have no idea why both cygwin and mingw can't use the standard
configure script without failing.

However, I finally got 0.9.7 no seg fault yay. Till today I never realized
that you can compile link and create an executable that will seg fault due
to mistakes in the configure script. However, when i think about it it
makes sense. the configure script sets up defines, and these defines
determine what is actually compiled.

I have definitely never had to worry about any of this before though. I'm
just a C# noob.

I separated out the part of the configure script that fails due to not
being able to remove conftest from the prior test and put it in its own "
testscript.in" file that i used to
generate a testscipt with autoconf tools. This then got me the correct
values of the 3 tests that failed in configure.

I then read through the configure script and found the names of the
libraries and define statements that should be placed in the makefile, and
added them to my environmental flags.

my successful flags to make up for configure not working for me were:
export CXXFLAGS="-DSIZEOF_UNSIGNED_LONG=4 -DHAVE_LIBREADLINE=1
-DHAVE_LIBHISTORY=1 -O1"
export CFLAGS="-DSIZEOF_UNSIGNED_LONG=4 -DHAVE_LIBREADLINE=1
-DHAVE_LIBHISTORY=1 -O1"

export LDFLAGS="-static-libgcc -static-libstdc++"
export LIBS="-lreadline -lhistory"

However, I have to mention that the -O1 was necessary to get it to build.
This was my fault for usinga  newer version of gperf. I was googling a bit
more for that gperf error and it actually came up in an old thread about
icarus.

https://groups.google.com/forum/#!topic/comp.lang.c++/qTic0RgVBxI

That thread points to a solution on
https://groups.google.com/forum/#!topic/comp.lang.c++/qTic0RgVBxI that
recommends either removing the inline attribute or adding -O1 to your
compile flags.

anyway as best I can understand it the gperf call would need to be updated
to some other syntax if it was desired to use newer gperf versions.

Next time i will be more careful to make sure that gperf matches the exact
version used in the build originally.

So I can now build 0.9.7 successfully. I can build 20150105 snapshot, but
that one has the segfault on unloading. and I still can't build the most
recent github master branch because of the ostream errors that i mentioned
earlier.

*5 minutes later*

one line change worked, and is successfully building. I can now use this
new build of icarus to have pausable simulation in windows and hook my
graphviewer onto it.

Thank you for all the help.

Btw, I noticed while trying to go back up through the post to find the line
that I needed to change that it somehow wasn't copied correctly in the
first post.

So again, in windows, if I change function signals_handler in
/vvp/schedule.cc by adding the underlined line.
extern "C" void signals_handler(int)
{
*      signal(SIGINT, &signals_handler);  *
      schedule_stopped_flag = true;
}

then I can pause and resume without any trouble from a windows terminal.  I
have no idea whether this works in linux or will break your other code.
Also, i note that in mingw, the console seems like it catches the signal
and doesn't pass it on to the program at all.  I assume linux at least is
nice enough to pass the signal on. As I understand it, linux sends the
signal to the program currently being run, while windows broadcasts the
signal to all programs in the terminal.

and like I said earlier, it seems windows deattaches the signal handler
when it receives the event, thus the need for the change.

I'm so glad to finally have it working, but I can't help but wonder, "who
could possibly keep track of all these little details necessary to get
things to work in different environments?"

But every day I see software that runs on linux and windows.

Anyway thanks a lot for your help and advice. I needed someone to help give
me tips and not go crazy.

Josh

caryr, 2015-05-01:
You are welcome. I am glad I was able to help and that you now have something that is working. We will look at the various issues and see what we can do to get things working as they should. We don't regularly compile for mingw so issues do come up from time to time in that environment. Truth be told, I have a virtual machine that runs Linux on my windows machine and that is what I usually use since it is significantly faster than cygwin and more closely matches my normal work environment.

Thank you for all the work and pointers to reports that describe the issues you are experiencing.

klammerj, 2015-05-04:
can you change signals_capture and signals_revert to use sigaction?

The Gnu manual (`man 2 signal`) sez:
```
[...]
   Portability
       The only portable use of signal() is to set a signal's disposition to SIG_DFL
       or SIG_IGN.  The semantics when using signal() to establish a signal  handler
       vary  across  systems (and POSIX.1 explicitly permits this variation); do not
       use it for this purpose.

       POSIX.1 solved the portability mess by specifying  sigaction(2),  which  pro‐
       vides explicit control of the semantics when a signal handler is invoked; use
       that interface instead of signal().
[...]
```

try this instead...
```
static struct sigaction prev_sa;
static void signals_capture(void)
{
      struct sigaction sa;
      memset(&sa,0,sizeof(sa));
      sa.sa_handler=&signals_handler;
      sigemptyset(&sa.sa_mask);
      sa.sa_flags = SA_RESTART;//for read()/write() etc...
      sigaction(SIGINT,&sa,&prev_sa);
}

static void signals_revert(void)
{
      sigaction(SIGINT,&prev_sa,NULL);
}

```

caryr, 2015-05-04:
Thanks for the suggestion. I will look into this.

knightpraetor, 2015-05-08:
Hey, I was wondering if there is a simple way that I can make icarus output
all the simulated vcd data.

It looks like the correct function is sys_dumpflush_calltf in sys_vcd.v

however, this seems to be getting stored in a bunch of structs buried
inside lots of vpi related files.  What's the best way for me to share
access to this function so that I can make schedule.cc dump all the data in
RAM into the vcd file when I pause the simulation?

Josh

caryr, 2015-05-08:
It's best to discuss this kind of thing on the development mailing list (sourceforge). It's also best to describe what you are trying to do. From what you have said this should already be happening because when you pause the simulation $fflush and $dumpflush are called by the stop handler and from the command prompt any system task should be callable.

knightpraetor, 2015-05-11:
ok. I subscribed to the dev list and if i have further issues I will post
there.  and I will take a look into what's happening in the stop handler.
Thanks again.

Josh

martinwhitaker, 2015-05-15:
The head of the master branch should now build OK if you use the mingw-w64 tools (either 32-bit or 64-bit). See http://iverilog.wikia.com/wiki/Installation_using_MSYS2 for details. I've tested this with Windows-8. Multiple ^C and cont are working for me when running vvp in a Windows command shell, but not in a MINGW shell (which I will investigate). If you have time, can you try following the build instructions on the Wiki and let me know if it works for you.

martinwhitaker, 2015-05-15:
Sorry, take that back about multiple ^C and cont. I'll try to fix it.

martinwhitaker, 2015-05-15:
Should be fixed now. We can't use klammerj's suggestion, because Windows doesn't provide sigaction.

caryr, 2015-05-15:
Once this is verified to work I think we should back port it to V0.9 since it is an obvious fix and we will likely do a final release of V0.9 that switches to using a suffix when we release V10.

martinwhitaker, 2015-06-22:
I'm going to close this, as I believe it is fixed. Please reopen it if the fix doesn't work for you.

closed 2015-06-22 by martinwhitaker