steveicarus/iverilog #65: Fails to compile

klammerj, 2015-04-30:
Just cloned the repository and tried to make
first with -j3
then without...
didn't work...
I believe it's related to the latest change to the makefile...

```
make[1]: Entering directory '/home/klammerj/projects/FPGA/src/ivl_ups_relly/tgt-pcb'
g++ -I. -I..  -DHAVE_CONFIG_H -fPIC -Wall -Wextra -Wshadow   -O0 -g -MD -c fp_lex.cc -o fp_lex.o
./fp.lex:27:18: fatal error: fp.h: No such file or directory
 # include  "fp.h"
 ```

caryr, 2015-04-30:
A fix for this should have been pushed. There was a build dependency that was missed.I will leave this open until it has been confirmed since this was working correctly for me.

klammerj, 2015-04-30:
It seems the fp.h rule does not trigger if the depfiles don't exist yet...
putting an explicit
`fp_lex.cc: fp.h`
in Makefile.in
works..

(try removing the dep directory, then running the build)

closed 2015-04-30 by klammerj