steveicarus/iverilog #68: RAM simulation mismatch

wluker, 2015-05-13:
When I describe a ram using two slightly different but valid descriptions and use the same testbench on both descriptions I recieve different results. I believe the results should be the same.

**file gold.v::**

    module bram_00_08(A1ADDR, A1DATA, B1ADDR, B1DATA, CLK1);
      input CLK1;
      input [0:0] A1ADDR;
      input [3:0] A1DATA;
      input [0:0] B1ADDR;
      output reg [3:0] B1DATA;
    
      reg [3:0] memory [0:1];
      initial begin
        memory[0] <= 4'd0;
        memory[1] <= 4'd0;
      end
    
      always @(posedge CLK1) begin
        memory[A1ADDR] = A1DATA;
        B1DATA <= memory[B1ADDR];
      end
    endmodule

**file gate.v::**

    module bram_00_08(A1ADDR, A1DATA, B1ADDR, B1DATA, CLK1);
      input CLK1;
      input [0:0] A1ADDR;
      input [3:0] A1DATA;
      input [0:0] B1ADDR;
      output reg [3:0] B1DATA;
    
      reg [3:0] memory [0:1];
      initial begin
        memory[0] <= 4'd0;
        memory[1] <= 4'd0;
      end
    
      always @(posedge CLK1)
        memory[A1ADDR] = A1DATA;
      always @(posedge CLK1)
        B1DATA <= memory[B1ADDR];
    endmodule

**file tb.v::**
Autogenerated using yosys.

    module testbench;
    
    integer i;
    
    reg [31:0] xorshift128_x = 123456789;
    reg [31:0] xorshift128_y = 362436069;
    reg [31:0] xorshift128_z = 521288629;
    reg [31:0] xorshift128_w = 1431531335; // <-- seed value
    reg [31:0] xorshift128_t;
    
    task xorshift128;
    begin
        xorshift128_t = xorshift128_x ^ (xorshift128_x << 11);
        xorshift128_x = xorshift128_y;
        xorshift128_y = xorshift128_z;
        xorshift128_z = xorshift128_w;
        xorshift128_w = xorshift128_w ^ (xorshift128_w >> 19) ^ xorshift128_t ^ (xorshift128_t >> 8);
    end
    endtask
    
    wire [3:0] sig_bram_00_08_B1DATA;
    reg [0:0] sig_bram_00_08_B1ADDR;
    reg [3:0] sig_bram_00_08_A1DATA;
    reg [0:0] sig_bram_00_08_A1ADDR;
    reg [0:0] sig_bram_00_08_CLK1;
    bram_00_08 uut_bram_00_08(
        .B1DATA(sig_bram_00_08_B1DATA),
        .B1ADDR(sig_bram_00_08_B1ADDR),
        .A1DATA(sig_bram_00_08_A1DATA),
        .A1ADDR(sig_bram_00_08_A1ADDR),
        .CLK1(sig_bram_00_08_CLK1)
    );
    
    task bram_00_08_reset;
    begin
        sig_bram_00_08_A1ADDR <= #2 0;
        sig_bram_00_08_A1DATA <= #4 0;
        sig_bram_00_08_B1ADDR <= #6 0;
        sig_bram_00_08_CLK1 <= #8 0;
        #100; sig_bram_00_08_CLK1 <= 1;
        #100; sig_bram_00_08_CLK1 <= 0;
        sig_bram_00_08_A1ADDR <= #2 ~0;
        sig_bram_00_08_A1DATA <= #4 ~0;
        sig_bram_00_08_B1ADDR <= #6 ~0;
        #100; sig_bram_00_08_CLK1 <= 1;
        #100; sig_bram_00_08_CLK1 <= 0;
    end
    endtask
    
    task bram_00_08_update_data;
    begin
        xorshift128;
        sig_bram_00_08_A1ADDR <= #2 { xorshift128_x, xorshift128_y, xorshift128_z, xorshift128_w };
        xorshift128;
        sig_bram_00_08_A1DATA <= #4 { xorshift128_x, xorshift128_y, xorshift128_z, xorshift128_w };
        xorshift128;
        sig_bram_00_08_B1ADDR <= #6 { xorshift128_x, xorshift128_y, xorshift128_z, xorshift128_w };
    end
    endtask
    
    task bram_00_08_update_clock;
    begin
        xorshift128;
        { sig_bram_00_08_CLK1 } = { sig_bram_00_08_CLK1 } ^ (1'b1 << (xorshift128_w % 2));
    end
    endtask
    
    task bram_00_08_print_status;
    begin
        $display("#OUT# %b %b %b %t %d", { sig_bram_00_08_A1ADDR, sig_bram_00_08_A1DATA, sig_bram_00_08_B1ADDR }, { sig_bram_00_08_CLK1 }, { sig_bram_00_08_B1DATA }, $time, i);
    end
    endtask
    
    task bram_00_08_print_header;
    begin
        $display("#OUT#");
        $display("#OUT#   A   sig_bram_00_08_A1ADDR");
        $display("#OUT#   B   sig_bram_00_08_A1DATA");
        $display("#OUT#   C   sig_bram_00_08_B1ADDR");
        $display("#OUT#   D   sig_bram_00_08_CLK1");
        $display("#OUT#   E   sig_bram_00_08_B1DATA");
        $display("#OUT#");
        $display("#OUT# A/--BC D /--E");
    end
    endtask
    
    task bram_00_08_test;
    begin
        $display("#OUT#\n#OUT# ==== bram_00_08 ====");
        bram_00_08_reset;
        for (i=0; i<1000; i=i+1) begin
            if (i % 20 == 0) bram_00_08_print_header;
            #100; bram_00_08_update_data;
            #100; bram_00_08_update_clock;
            #100; bram_00_08_print_status;
        end
    end
    endtask
    
    initial begin
        // $dumpfile("testbench.vcd");
        // $dumpvars(0, testbench);
        bram_00_08_test;
        $finish;
    end
    
    endmodule

Thank you for a nice tool!

steveicarus, 2015-05-13:
Your models are NOT the same. There is no defined order that
triggered statements are executed, so your gate.v version has
a race in it, which will be evident when A1ADDR and B1ADDR are
equal. Your gold.v example will output B1DATA==A1DATA when the
addresses are the same, your gate.v version has less well
defined (due to races) results.

On 05/13/2015 09:09 AM, wluker wrote:
> When I describe a ram using two slightly different but valid
> descriptions and use the same testbench on both descriptions I
> recieve different results. I believe the results should be the
> same. [...]

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

wluker, 2015-05-13:
Thanks Steve! I just saw that too and was writing to close this.

Ackkk... I just realized this as well, but you were faster. When I use the proper nonblocking assigments, it works as expected. 
This was a greatly simplified test case in yosys that was giving me trouble, so I need to send a fix to the yosys project.
Thanks again.

closed 2015-05-13 by wluker