steveicarus/iverilog #73: vpi: tri bus split

themperek, 2015-06-30:
Something that worked last week does not work now. 
Works with Modelsim as expected.
When tri state bus is split the state is not propagated.

```cpp

#include <assert.h>
#include "vpi_user.h"

static PLI_INT32
EndOfCompile(s_cb_data *data)
{
    s_vpi_time timerec = { vpiSimTime, 0, 0, 0 };
    s_vpi_value val_b0;

    vpiHandle bd_handle;

    (void)data;

    bd_handle = vpi_handle_by_name("test.BD", 0);
    assert(bd_handle);

    val_b0.value.str = "zzzzzzzz";
    val_b0.format = vpiBinStrVal;
    vpi_put_value(bd_handle, &val_b0,  &timerec, vpiInertialDelay);
    
    timerec.low = 1000;
    val_b0.value.str = "00000010";
    vpi_put_value(bd_handle, &val_b0,  &timerec, vpiInertialDelay);
    
    timerec.low = 2000;
    val_b0.value.str = "zzzzzzzz";
    vpi_put_value(bd_handle, &val_b0,  &timerec, vpiInertialDelay);
    
    timerec.low = 3000;
    vpi_put_value(bd_handle, &val_b0,  &timerec, vpiInertialDelay);
    
    return 0;
}

static void
VPIRegister(void)
{
    s_cb_data cb_data;
    s_vpi_time timerec = { vpiSuppressTime, 0, 0, 0 };

    cb_data.time = &timerec;
    cb_data.value = 0;
    cb_data.user_data = 0;
    cb_data.obj = 0;
    cb_data.reason = cbEndOfCompile;
    cb_data.cb_rtn = EndOfCompile;

    vpi_register_cb(&cb_data);
}

void (*vlog_startup_routines[]) () = { VPIRegister, 0};
```

```verilog
`timescale 1 ns / 1 ns

module bus_to_ip
#(
    parameter DBUSWIDTH = 8
)
(
    inout wire [DBUSWIDTH-1:0] BUS_DATA
);

assign BUS_DATA = {DBUSWIDTH{1'bz}};

endmodule

module test(
    inout wire [7:0] BD
);
    
    bus_to_ip #( .DBUSWIDTH(4)) i_bus_to_ip1
    (
        .BUS_DATA(BD[3:0])
    );
    
    bus_to_ip #( .DBUSWIDTH(8)) i_bus_to_ip2
    (
        .BUS_DATA(BD)
    );
    
    initial begin
        $dumpfile("testz.vcd");
        $dumpvars(0);
    end
    
    initial begin
        #1500
        $display("BUS_IP1 = %b BUS_IP2 = %b @ %0t", i_bus_to_ip1.BUS_DATA, i_bus_to_ip2.BUS_DATA, $time);
    end
    
endmodule

```

Output:
```
BUS_IP1 = zzzz BUS_IP2 = 00000010 @ 1500
```

martinwhitaker, 2015-06-30:
I've pushed a fix for this. I still need to do a bit more work to make force/release work properly in this case.

martinwhitaker, 2015-07-01:
Fixed the VPI release behaviour in this case.

closed 2015-07-01 by martinwhitaker

themperek, 2015-07-02:
Thank you.
Works: https://travis-ci.org/SiLab-Bonn/basil