steveicarus/iverilog #78: Auto-detection of version number

xyproto, 2015-08-25:
Hello,

I'm currently maintaining the [iverilog](https://www.archlinux.org/packages/?q=iverilog) package for Arch Linux. In that connection, I wish to be able to automatically detect the latest version number of packages, preferably be reading an HTTP page, by using a small commandline utility I have written.

However...
* The main web page does not contain the current version number.
* I found this file, but the version number seems to be out of date (and it's FTP, not HTTP): ftp://ftp.icarus.com/pub/eda/verilog/snapshots/README
* version_base.h and README.txt unfortunately contains the next version number, not the current one.

Could you please add "10.0" somewhere in the README.txt file, or on http://iverilog.icarus.com/?

Thanks.

Best regards,
Alexander F Rødseth / "xyproto"

steveicarus, 2015-08-25:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Hmm...

I've put a README in the ftp://ftp.icarus.com/pub/eda/verilog
directory that points you to the v10 directory. I guess from
there you can note the latest file that matches verilog-10.*.tar.gz.
Will that work for you?

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlXcpGQACgkQrPt1Sc2b3iksdgCeJO4cfBpIuK9zODG2IIbd+G/0
C9gAoNHObQj5dOEJeSc8SBUy3j44KGmm
=suNS
-----END PGP SIGNATURE-----

xyproto, 2015-08-25:
Any document accessible by HTTP that states that the latest version is 10.0 would work (or "v10", if that is the version number). This is just a "nice to have" feature request, though, so no need if it is cumbersome to implement or maintain.

caryr, 2015-08-25:
The snapshots page does not contain released version. The version_base.h and README.txt files should have the correct information for the given version. Note that github does have the next release information since that is what it is targeted at.

Debian has a watch file that they use and while I have not verified that it works with our recent change to 10.0 it sure looks like it should since 10.0 is greater than 0.9.

I will upload the V10 file to SourceForge and created a stub release notes page on the wiki.

themperek, 2015-08-26:
Would not be nice/better to put various binaries under github releases?
https://github.com/steveicarus/iverilog/releases

steveicarus, 2015-08-26:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I can look into that.

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlXd3UIACgkQrPt1Sc2b3ilTrQCfZOLT68BGw9FkXZ2jXij3knnZ
3W0AnRr4E/T3YU151rBqqn/UAtwvWQRE
=/+Rx
-----END PGP SIGNATURE-----

steveicarus, 2015-10-01:
I think it is time to close this. I do not want to put binaries or tarballs in the git repository, the tags along with github extraction tools, serve that purpose well enough and we have other places (enough places!) for storing binaries and packages.

closed 2015-10-01 by steveicarus

xyproto, 2015-10-01:
What is the latest version number? v10, v10_0, v10.0 or 10.0?

steveicarus, 2015-10-01:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

The version is 10.0, the tag that identifies its place in the
repository is v10_0. The 10.* releases are on the v10-branch
git branch.

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlYNr8kACgkQrPt1Sc2b3imFhACgq3M3WHBi9Ez2HML4AMbeHFKT
/pIAoJWuQ+LdDiB2ldR1UlUTURjYCG3p
=q4V8
-----END PGP SIGNATURE-----