steveicarus/iverilog #80: Unexpected behaviour of events edge triggered on register value

sultanqasim, 2015-10-02:
I'm not sure if this is a "feature" or a bug. Be easy on me, I'm new to Verilog, so maybe my understanding is faulty.

From what I understand, when you set a register's value multiple times in an always at edge block, the register should only take on the last value it was assigned, and not glitch to the values you assigned to it earlier in the block before reaching the final value. This would match the hardware behaviour of a multiplexed input (combinational logic) being fed to a flip flop (register).

I had some code where I had a register that acted as a clock for some other modules that were edge triggered off it. My original usage was more complicated, but my simplified test case is this:
```
reg cclk;
reg override;

...

always @(posedge clk) begin
    cclk <= 0;
    if (override) cclk <= 1;
end
```

In this code, I expect cclk to remain high after every clk edge if override is held high, and I don't think cclk should glitch to zero for an instant on every clk edge. However, Icarus is modelling this as cclk momentarily glitching to zero for a zero length period on every clk edge. This causes events on cclk edges to be triggered.

Changing the cclk driving code to this avoids the issue:
```
always @(posedge clk) begin
    if (override) cclk <= 1;
    else cclk <= 1;
end
```

See my full test case https://gist.github.com/sultanqasim/44b0b5cf8dbf9152ac6e and screenshot of the result http://i.imgur.com/b7DJrqh.png

Is my understanding of the expected behaviour correct (ie. is this a bug?), or is this a case where the simulation is supposed to differ from reality?

martinwhitaker, 2015-10-02:
No, your understanding is incorrect. The relevant section of the IEEE standard states:

"Nonblocking assignments shall be performed in the order the statements were executed (see 9.2.2). Consider the following example:

    initial begin
      a <= 0;
      a <= 1;
    end

When this block is executed, there will be two events added to the nonblocking assign update queue. The previous rule requires that they be entered on the queue in source order; this rule requires that they be taken from the queue and performed in source order as well. Hence, at the end of simulation time 0, the variable a will be assigned 0 and then 1."

It's not really fair to say that in this case simulation differs from reality. Your original code example doesn't properly describe a flip-flop with a mux input, but your revised example does.

closed 2015-10-02 by martinwhitaker

sultanqasim, 2015-10-02:
Thanks for the quick and clear response