steveicarus/iverilog #85: Possible race condition with `make install`

dunn, 2015-11-30:
Running `configure && make install` usually works, but on one occasion I got this error, which looks like it might be a race condition: https://gist.github.com/dunn/7e611d870ab56af1622c#file-icarus-verilog-log-L396-L397 (system configuration included)

Since it's only occurred once, I'm not sure exactly what's going on or how serious it is.

steveicarus, 2015-12-01:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I'll look into this as soon as I can. I'm kinda busy right now
with day-job stuff.

> Running |configure && make install| usually works, but on one
> occasion I got this error, which looks like it might be a race
> condition: 
> https://gist.github.com/dunn/7e611d870ab56af1622c#file-icarus-verilog-log-L396-L397
(system configuration included)
> 
> Since it's only occurred once, I'm not sure exactly what's going on
> or how serious it is.

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlZeARgACgkQrPt1Sc2b3ik7wgCgyBro91eMpWqHKdl69R62cZJV
xIIAn3g69/m5xu/8l7mFtUUPmbUIqr7D
=/cCc
-----END PGP SIGNATURE-----

martinwhitaker, 2019-07-25:
Has this ever occurred again? I have never seen it in many, many builds.

dunn, 2019-07-25:
No idea; I haven't touched this in a long time.

martinwhitaker, 2019-07-26:
OK, closing as unreproduceable.

closed 2019-07-26 by martinwhitaker