steveicarus/iverilog #87: What's the proper name of this software?

krytarowski, 2015-12-13:
Icarus Verilog was used to be packaged in pkgsrc for NetBSD as "verilog" and "verilog-current". Later it was repackaged from GitHub as "iverilog". While some people argue that is might be called "icarus-verilog" or "icarus".

Please unify the naming and names of tarballs:

http://iverilog.wikia.com/wiki/Installation_Guide

We hold on with changes in pkgsrc till the naming will be solved. Another issue is that whether it's sane to package -current snapshots?

steveicarus, 2015-12-13:
The tarballs and packages are conventionally named "verilog-*", and the binary programs are
iverilog, etc. We have no control over how 3rd party packagers name the packages.

krytarowski, 2015-12-13:
Could we please unify it to reduce confusion? If I was aware that 'verilog' is Icarus, I wouldn't repackage it as 'iverilog'.

steveicarus, 2015-12-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

On 12/13/2015 02:53 PM, krytarowski wrote:
> Could we please unify it to reduce confusion? If I was aware that 
> 'verilog' is Icarus, I wouldn't repackage it as 'iverilog'.

Again, I have no control over it. We always call the package,
whether tarball or rpm, "verilog-<version>" and that is that.
What the netbsd or freebsd people do is out of our control.

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlZt+KQACgkQrPt1Sc2b3imDcwCgm3Z3OoxQq9bKyOQWDAvtQVMG
ewAAn3mcOdw8mwino6bH5P/unCQoVo8l
=YkfP
-----END PGP SIGNATURE-----

krytarowski, 2015-12-13:
But I'm talking about upstream part:

```
% tar -zxvf verilog-version.tar.gz
% cd verilog-version
```

vs

```
% git clone git://github.com/steveicarus/iverilog.git
% cd iverilog
```

There is 'verilog' or 'iverilog' depends of the source of distribution. It makes confusion for 3rd party redistributors how to handle it properly.

krytarowski, 2015-12-13:
If it was always by convention 'verilog' it might be worth to name the git repo the same way.

Personally I would prefer the 'iverilog' name, since it's a suite for the verilog language.

Thank you for your software!

gdt, 2015-12-14:
I'm also involved in packaging in pkgsrc.  We get it that you can't control what we do, but that's not the point; we are trying to follow upstream's naming.   From the pkgsrc point of view -- and I am pretty sure other packaging systems see it the same way -- each package should have a particular name.  That name should be consistent across packaging systems, so that people can find the package.  And the package name (which we choose), the tarball name, the directory name the tarball unpacks to, and the name as described on web pages should all be the same.  By default, pkgsrc expects the package name, the distfile name, and the unpack directory to match, and for most packages, this is true.  This is really a simple plea for consistency.

As I see it there are multiple names in play
  - verilog (distfile name, unpack name for actual release)
  - iverilog (program name as installed, which is ok to differ, and github repo name)
  - "Icarus Verilog" long name as described in text on the web
  - "Icarus" shorter name also used on the web

So basically the requests is for you to decide on calling the package-name/tarball-name/unpack-directory either verilog, or iverilog, or icarus-verilog, or icarus, but to pick one and be consistent.  As an example, using the English name "Icarus Verilog" and the distfile name, unpack name, and github repo name "iverilog" would be one such consistent choice.    Really, this bug report was filed because the github name and the installed program are iverilog and the distifle and unpack name are verilog.

caryr, 2015-12-14:
My assumption is if we want consistency it should be iverilog. Debian has a verilog transition package though iverilog looks to be the main package name. I'm assuming it would not be a problem if we changed the tar file name for them. I think if we get a reasonable proposal we can do something that is consistent, but we need input from all people who package iverilog before we would consider a change (well maybe most and all the majors would be enough). Like Steve said this is a packaging issue and we only provide source in either tar or git format. Unfortunately we have some inconsistency between these two formats, but we do not want to break people who are already using this inconsistent naming system.

krytarowski, 2015-12-14:
http://pkgs.org/search/verilog shows me:

Mageia: iverilog
Slackware: verilog
Fedora (CentOS/RHEL): iverilog
openSUSE (SUSE): iverilog
Debian: iverilog and legacy named package verilog
ArchLinux: iverilog
ALT Linux: iverilog

It looks like Debian switched lately from verilog to iverilog, while other are already there. Legacy distros like Slackware keep old names.

pkgsrc (NetBSD) still has verilog.

caryr, 2015-12-14:
Yes, iverilog looks like the right package name. The issue is will everyone buy into us changing the tar file from verilog to iverilog?

krytarowski, 2015-12-14:
As a packager with experience with RPM packages and in the pkgsrc world I see no issues.

Furthermore, renaming tarballs to iverilog will make life easier. Usually packaging systems are prepared for this case when tarball name = package name and perform default actions out of the box, while customized setups need more scripting.

caryr, 2015-12-14:
Are the other people who are already looking for verilog-* going to be happy is we change to iverilog-*. More specifically they will have to change some code that checks for version updates, etc. I agree consistency is good, it's just that we have been doing things wrong for a long time so we don't want to break things for people unless they agree it's okay. I also find it a real hack to provide both files which could lead to even more confusion.

krytarowski, 2015-12-14:
That's a philosophical question. Not a single change is without making upset someone. We need to bear with it.

steveicarus, 2015-12-14:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I'm a little nervous about creating even more chaos then already
exists. The current situation may not be ideal, but it is stable.
It is *15*years* stable. We can make a pronouncement about the
current state of affairs so that 3rd parties better follow the
convention, but if we decide to make some name changes, we'll
need to take this discussion to the iverilog-devel mailing list.

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlZvFisACgkQrPt1Sc2b3ilXcgCdE4j67jJU1WyrX5Yc1SajXzlE
ZP4An2C15cIeGjfphvgFO40y7B0gELBS
=y+vc
-----END PGP SIGNATURE-----

gdt, 2015-12-15:
@steveicarus: Fair enough on stability.  a few thoughts:
  - Given the mismatch of names, the project should declare a preferred package name.  I know you may feel you lack the authority to give orders to packagers, but generally we're happy to follow upstream's (sane) guidance, and it's better for all of our users to be consistent.
  - any name mismatches from the preferred name (e.g. verilog in source tarball) should at least be acknowledged as a bug that isn't being fixed to avoid churn
  - I suspect very few people other than packagers, perhaps zero people, care about the name having slight changes.   Making the next source release iverilog- and unifying on that would probably be a net win when summed over all humans.

steveicarus, 2015-12-15:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I'm perfectly happy with this, I was going to suggest essentially
the same thing.

We do have a convention:

  * package name is "verilog"
  * The tarball is called verilog-<version>... and extracts into
    the directory "verilog-<version>/"
  * The installed command line programs are "iverilog", "iverilog-vpi",
    etc. See the bundled verilog.spec file with source distributions.

On 12/14/2015 04:57 PM, Greg Troxel wrote:
> @steveicarus <https://github.com/steveicarus>: Fair enough on
> stability. a few thoughts:
>
> * Given the mismatch of names, the project should declare a
> preferred package name. I know you may feel you [...]

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlZvZwYACgkQrPt1Sc2b3ikQmACgtfO44rNeo704mLtyQfaOUU0c
IucAnRdIda5OO3Qw5GkjAZGocXru9DZs
=O8fe
-----END PGP SIGNATURE-----

krytarowski, 2015-12-15:
OK, so as I understand 'verilog' stands.

steveicarus, 2015-12-15:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

On 12/14/2015 05:06 PM, krytarowski wrote:
> OK, so as I understand 'verilog' stands.

Yes.

Given what I know now, 17 years later, I might have done it
differently (i.e. called everything "iverilog") because calling
it "verilog" is slightly presumptuous; but the convention is what
it is now, it doesn't conflict with anybody else, and changing
it now would make even more of a mess.

So, that's my ruling.

Suggestions for where I should write it down, so that folks like
you would notice it?

- -- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iEYEARECAAYFAlZva+cACgkQrPt1Sc2b3imnygCgzm5wSf5cGnmo3fGGukTT6gaQ
TzcAoMWP9CQchygngHLZCXWlSnYMojtF
=y8mt
-----END PGP SIGNATURE-----

krytarowski, 2016-10-08:
I've renamed it to iverilog in pkgsrc.

closed 2016-10-08 by krytarowski