steveicarus/iverilog #93: -Wimplicit-dimensions and output reg

ravenexp, 2016-03-15:
I have a simple testbench module approximately as follows:

```
module delay_model
        (
         input  d_i,
         output reg d_o
         ); 

        always @(d_i) #1 d_o = d_i;
endmodule
```

Starting with 10.1 when compiled with -Wall I get:
> warning: Port declaration of d_o inherits dimensions from var/net.

What is the cause of this?

steveicarus, 2016-03-15:
Looks like a bug in the warning. It is supposed to warn if you declared a variable or net as a vector
(i.e. using the [x:y] syntax) and as a scaler some place else. In Verilog, that means that the vector
form is taken, but Xilinx xst does this wrong, so I added the warning so that Xilinx users can fix
their code.

Since I added this feature (and the bug) I guess it is up to me to fix it.

martinwhitaker, 2019-07-25:
This is fixed in v10.2.

closed 2019-07-25 by martinwhitaker