steveicarus/iverilog #94: Parameter used in assignment sizing

hammy025, 2016-03-23:
I've looked through a lot of bugs/issues and didn't see this exact one, but please correct me if this is asked and answered already.

Also, I can't think of any cases off the top of my head where this is functionally required.  It is normally done (in my experience) simply to satisfy Spyglass or other linting tools by keeping LHS/RHS widths equal.

The issue I am seeing is that a parameter cannot be used when sizing the RHS of an assignment.  For instance:
`dout = WIDTH'(din + 1);`
where WIDTH is a parameter and I want to assign 'WIDTH bits' of the expression into dout.  This works fine if the parameter is replaced by a constant:
`dout = 32'(din + 1);`

My understanding of parameters in 1364-2005 is that they are treated as constants after elaboration.  1364-2005, 4.10.1:

> Parameters represent constants; hence, it is illegal to modify their value at run time. However, module parameters can be modified at compilation time to have values that are different from those specified in the declaration assignment. This allows customization of module instances. A parameter can be modified with the defparam statement or in the module instance statement. Typical uses of parameters are to specify delays and width of variables. See 12.2 for details on parameter value assignment.

If that's the case, WIDTH'(din + 1) would be a constant size and no size casting is performed.

I'm fairly new to this level of detail of the spec, so I may be way off.
I ran the below code with "iverilog -g2012 param_sizing_fail.v".
Icarus Verilog version 10.0 (stable) (v10_0)
Errors:
> param_sizing_fail.v:10: syntax error
> param_sizing_fail.v:10: error: malformed statement

param_sizing_fail.v contains:
```
module param_sizing_fail (
  output logic [WIDTH-1:0]  dout,

  input  wire  [WIDTH-1:0]  din
  );

  parameter            WIDTH = 32;

  always @(*) begin
    dout = WIDTH'(din + 1);
  end

endmodule
```

martinwhitaker, 2016-03-24:
Yes, currently Icarus only supports size casting when the size is a literal value. The implementation is also a bit broken, as it doesn't properly calculate the resulting expression type and width. I'm working on a fix for this.

hammy025, 2016-03-25:
Thank you thank you ^.^

martinwhitaker, 2016-03-25:
Fixed in development and back-ported to v10 branch (as it fixes bugs in the existing implementation).

The vlog95 target needs a bit of work to support this.

hammy025, 2016-03-26:
It's working for me on the v10 branch.  Thank you SO much!
I would close it, but not sure if you wanted to leave it open for the vlog95 target.

martinwhitaker, 2016-03-26:
I've had a look at fixing the vlog95 target, but can't think of a way to implement the required functionality in standard Verilog, so I've added the failing test to the list of expected failures that need further investigation in regress-vlog95.list. As the failing code uses signed values, it can't be converted to pure Verilog-95 anyway, so I think we can close this issue now.

closed 2016-03-26 by martinwhitaker