steveicarus/iverilog #97: System verilog Port Declaration Problem

red0bear, 2016-03-31:
Hello people !  I would like thank you guys to your contribution in icarus verilog program is very nice ! im alredy working with it using pli/vpi mixed with systemC and it works well. Sure some times i get troubles but its ok. 

My problem resume i need simulate a IP coded in system verilog and when i try run it , show syntax errors. I have looked on source code and i see this kind of declaration is not accepted in module.

example:
output logic [2:0] 	CURRENTSTATE;

Obs: logic declared inside modules is accepted.

Another problem i have seen is always_ff is not accepted "not sure about this". 

I know across Source Forge forum system verilog inst complete yet but is good show this to know if is a problem or not.

martinwhitaker, 2016-04-01:
This is the same issue as https://sourceforge.net/p/iverilog/bugs/974.

always_ff, always_comb, and always_latch are not yet supported.

red0bear, 2016-04-01:
Hummmmm ... anyway if i try use always(....) or always(*) instead always_ff/always_comb could change something ? 

I m so sorry to repeat bug but i wish sometimes icarus verilog have a good manual notice about this facts. I know about wiki ... but inst enough to cover all features till moment done.

martinwhitaker, 2016-04-01:
I agree, it would be really useful to have a table of supported features on the Wiki. I don't exactly know what is and isn't supported myself.

martinwhitaker, 2016-04-04:
I've added support for SV types in non-ansi style port declarations in the master branch, so am closing this issue.

closed 2016-04-04 by martinwhitaker