steveicarus/iverilog #98: Inheritance core dump

hammy025, 2016-04-04:
I thought perhaps inheritance wasn't supported, but looking at the source code, it looks like it should be.  This seems like a very straight forward example so maybe I'm just missing something or made some stupid mistake.  If so, sorry in advance.

It core dumps after an ivl assertion while scoping the derived class.  The error message is:

> /code/Hardware/verif/object/object.sv:27: assert: elab_scope.cc:503: failed assertion scope
> Aborted (core dumped)

Line 27 is:
`class test extends object;`

Class declarations:
```
class object;
  string Type;

  function new(string iName="object");
  begin
    Type = "object";
  end 
  endfunction
endclass

class test extends object;

  function new(string iName="test");
  begin
    super.new(iName);
    Type = {Type, ".test"};
  end 
  endfunction
endclass
```

Stimulus code:
```
module top();

  object obj;
  test tst;

  initial begin
    obj = new("obj");
    $display("obj type is '%s'", obj.Type);
    tst = new("tst");
    $display("tst type is '%s'", tst.Type);
  end 
endmodule
```

steveicarus, 2016-05-03:
The basics of class derivation do work. You can see working
examples in the ivtest suite in sv_class20-23.v examples. So
there is more to it then just class derivation. Can you post
a complete example that gives you this error?

hammy025, 2016-05-03:
Files are attached.  Command line is simply:
iverilog -g2012 object.sv top.sv

[inheritance_files.zip](https://github.com/steveicarus/iverilog/files/246229/inheritance_files.zip)

steveicarus, 2016-05-03:
Icarus Verilog seems to be not handling the case that all the classes are in the $root scope. I'll put some effort into this.

martinwhitaker, 2017-10-08:
This should now be fixed in both the master and v10 branches.

closed 2017-10-08 by martinwhitaker