steveicarus/iverilog #99: Run time error? internal error

aolofsson, 2016-04-04:
Hi,

Huge fan of the tool, use it every day!

I write a lot of parametrized modules and sometimes I make vector sizing mistakes that only get caught at runtime.  Any workaround, compiler switch to give more insight? I tried -Wall, but doesn't seem to provide the anything missing. Verilator runs through clean.  Currently I have to resort to divide and conquer and code review to find these bugs. (very painful!)

Version: 10.1

Error: "internal error: 10vvp_cmp_eq: recv_vec4_pv(6'b000000, 0, 6, 10) not implemented"

aolofsson, 2016-04-04:
fyi: found the bug in my code (4 hrs later through divide and conquer:-)...) Line goes into some arithmetic logic..(sorry can't share the complete test case). Can imagine the internal error having something to do with arithmetic comparison "a>b" or "a==b"...

Bug in code:

was:          assign plane_id[5:0]    = 10'b0;

should be: assign plane_id[9:0]    = 10'b0;

martinwhitaker, 2016-04-04:
Thanks for reporting this. I've reproduced the bug with this simple test case:
```
module test();

wire [9:0] value1;
wire [9:0] value2;

assign value1[5:0] = 10'b0;

assign value2 = value1 + 10'b1;

endmodule
```
This is a bug in the compiler or runtime, as this is perfectly legal Verilog (even if it wasn't, the compiler should report an error, not crash at runtime).

In regard to tracking down bugs, if the problem occurs in procedural code, compiling with the option '-pfileline=1' will cause the runtime to report the file/line of each statement it executes. This is very verbose, but can be helpful. For bugs in continuous assignments though, I know of no better solution than divide and conquer.

aolofsson, 2016-04-04:
Thanks for the reply!Look forward the fix. I have only had 3 of these bugs in the last year, so it's not too bad but it hurts every time.:-)

martinwhitaker, 2016-04-04:
I've pushed a fix for this particular case to both the master and the v10 branches. But there are some similar issues with other functors that need to be fixed as well.

martinwhitaker, 2016-05-08:
This should now be fully fixed.

closed 2016-05-08 by martinwhitaker