steveicarus/iverilog #101: testBench declaration ghost ports

red0bear, 2016-04-06:
Hi again people.

i was building the top module using verilog and forgot change to correct ports and icarus compiled it without errors. The follow example : 

module TopTestbench;

  reg a,b,c,d;

  TESTE DUT(
                       .a(a),
                       .b(b),
                       .c(c_i),
                       .d(d_i)
                      );
endmodule

it take normal compilation using iverilog , but when i try set or get value using VPI he show error.

steveicarus, 2016-04-06:
If I understand what you are trying to say, then this is not a bug. Verilog allows named wires
to be created implicitly. (Yuck!) Add this to your source file and you will get useful error
messages

`default_nettype none

red0bear, 2016-04-06:
I was suspecting about this. But in all case i write to if is really this. On my case i was trying how i use VPI to access variables i give me a error when i try write / read values using vpi_get_value / vpi_put_value. So i presume vpi is not seeing these implicity variables.

martinwhitaker, 2016-05-08:
If I understand you correctly, the attached example (adapted from the force_wire test in the test suite) does what you want.

[force_implicit.zip](https://github.com/steveicarus/iverilog/files/254106/force_implicit.zip)

red0bear, 2016-05-09:
I understand now what should be made. i use direct declaration:

Example: https://github.com/GLADICOS/UART/blob/master/vpi/execute_uart.h

On your example you use with iterators, that’s nice; but i need make explicit declaration to show people what i'm exactly doing in task.  But i see there is another way to do like you show, but i tried call a implicit net using a root path.

martinwhitaker, 2016-05-11:
Direct reference also appears to work - see revised example:

[force_implicit.zip](https://github.com/steveicarus/iverilog/files/258818/force_implicit.zip)

red0bear, 2016-05-11:
On first post where i did this mention .... i just tried with : 

TESTE DUT(
.a(a),
.b(b),
.c(c_i), <-- c_i is not declared 
.d(d_i) <-- d_i is not declared
);

Your example works fines .... i think the problem is it don't see implicty declared when you make instance DUT. Yor test you

martinwhitaker, 2016-05-11:
IN and OUT are implicitly declared in my example.

If you still think there's a problem, please provide a complete test case ('C' code as well as Verilog) we can look at.

red0bear, 2016-05-12:
Martin, 
  
   i will explain better ; the examples where you provide works very fine. There is no mistakes now. I had already made git pull to recent branch. But i need apologize you cause i didn't tested again, i discovered this because i forget to change names on top level. But i will use your example to show you the bug is gone. You tested with two approaches , this on annex is where i found this problem. Here it run clear for me now. 

i just changed to the form where i found the problem -> dut dut(.i(IN), .o(OUT)); 

i think you can close this topic about this. :+1: 

[force_implicit.zip](https://github.com/steveicarus/iverilog/files/260499/force_implicit.zip)

red0bear, 2016-05-12:
And thank you to correct the thing :D

martinwhitaker, 2016-05-13:
Good to hear your issue is resolved. I'll close this now.

closed 2016-05-13 by martinwhitaker