steveicarus/iverilog #102: Include parameters using packages

red0bear, 2016-04-26:
Hello people, 

Im trying to use include package on system verilog where have group of parameters. Example

name file: parameters.sv
 
package TEST_PARAMETERS;

   parameter GND           = 64'h00;
   parameter ALL_ZERO = 64'h00;
   parameter ALL_ONE = 64'hFF;

endpackage

name file : test_case.v / test_case_1.sv

import TEST_PARAMETERS::*;

module Mytest(....);

endmodule

When i try call using import i get a syntax error "I give up". There some support to imports on icarus to System Verilog ?

steveicarus, 2016-04-28:
You use the -g2012 or similar compiler flag to enable System Verilog?
There is some import (and package) support in Icarus Verilog. If you
can attach some actual files, we can look into this more deeply.

red0bear, 2016-04-28:
Well , in this case example is more near with what im using. I will try take a better example. The files where i have now i cant share.

red0bear, 2016-04-28:
This is simple example using a parameter to a fifo example, in previous message i tried use with -g2012 this flag but still giving the error. I hope example help. 

To see what happen

I tryed command : iverilog -g2012 *.sv  -> but still giving the same error

[rtl.tar.gz](https://github.com/steveicarus/iverilog/files/240460/rtl.tar.gz)

steveicarus, 2016-04-28:
Looks like the problem is the "import ...;" in the $root scope.
In fact, I don't even think that is allowed in the IEEE1800 standard.
Can someone point me to where I'm wrong about that?

You can get what you want by moving the import into the module, and
using fully qualified references in the port list. Like so:

module sync_fifo
(
	input CLK,
	input RST,
	input [FIFO_PARAMETERS::FIFO_WIDTH-1:0] WR_DATA,
	input WR_ENA,
	input RD_ENA,
	output [FIFO_PARAMETERS::FIFO_WIDTH-1:0] RD_DATA,
	output FULL,
	output EMPTY
);

   import FIFO_PARAMETERS::*;
   [...]
endmodule

red0bear, 2016-04-28:
I tested the file on Cadence and it Works, and the another files where i'm trying to run your simulator. But i don't know if this information could help.

steveicarus, 2016-04-28:
It's interesting, thanks. I've looked at the formal syntax
and I just don't see where "import" keywords can be used in
the $root scope. It's possible that it is there and I'm just
missing it, and it is also possible that Cadence is generous
with its syntax. If other SystemVerilog compilers handle the
syntax, then it's kinda moot:-/

red0bear, 2016-04-28:
i dont know what could be steve. But i hope you decide what is the best to do: :-)))))))). i'm here waiting to make git pull :+1:

martinwhitaker, 2016-04-29:
I think this is an import into the $unit scope, which the IEEE standard does allow, by:

source_text -> description -> package_item -> package_or_generate_item_declaration -> data_declaration -> package_import_declaration

There's a sentence in section 23.11 that backs this up:

"Declarations present or imported into $unit are not visible in the bind statement."

steveicarus, 2016-04-29:
Good enough for me. I'll look into implementing it.

svenka3, 2017-10-04:
Steve - I hit the same issue today with IVL. By any chance do you've a version with this support? 

Thanks
Srini

steveicarus, 2017-10-04:
I don't remember doing anything with the "$unit" scope. The "$root" scope is
supported, but that's different. I don't even know what $unit is.

martinwhitaker, 2017-10-04:
The $unit scope is the compilation unit (i.e. file). See also the preceding comments on this issue (it was a while ago ;-) ).

martinwhitaker, 2017-10-09:
Having read the standard a bit more closely, I'll amend my previous comment. The standard says

> compilation unit: A collection of one or more SystemVerilog source files compiled together.
> 
> The exact mechanism for defining which files constitute a compilation unit is tool-specific. However, compliant tools shall provide use models that allow both of the following cases:
> 
> a) All files on a given compilation command line make a single compilation unit (in which case the declarations within those files are accessible following normal visibility rules throughout the entire set of files).
> 
> b) Each file is a separate compilation unit (in which case the declarations in each compilation-unit scope are accessible only within its corresponding file).
> 
> The contents of files included using one or more `include directives become part of the compilation unit of the file within which they are included.
>
> If there is a declaration that is incomplete at the end of a file, then the compilation unit including that file will extend through each successive file until there are no incomplete declarations at the end of the group of files.

Yuck! Didn't the standards committee learn anything from `timescale and `define ?

Icarus currently implements option (a) but not option (b). I'll take a look at fixing that, as (b) is the sane choice.

tovine, 2018-03-14:
I have the same problem when trying to compile and simulate the FPU from the pulp-platform: https://github.com/pulp-platform/fpu

Tried to use the newest version from `master` today (5cfb7d68ae9482b8d73365f64b47baf33a61f767), as you mentioned fixing something possibly related in this issue: https://github.com/steveicarus/iverilog/issues/183?
I too get 
```
fpu/hdl/fpu_v0.1/fpu_private.sv:34: syntax error
I give up.
```
That line has the `import fpu_defs::*;` statement, outside the `module` scope (but should be inside the `$unit` scope if I understand correctly)

wallento, 2018-04-30:
Hi,

I have stumbled over this issue repeatedly. I don't think `$root` and `$unit` are relevant in this context. A.1.2 applies, saying that a source text is an optional timescale and then zero or more of the various declarations we know. The relevant syntax item is `package_import_declaration` and this is not allowed in this context.

I think the fact that all other tools I know support this syntax may anyways be an argument to support it too, generally. Anyhow, the context of the compilation unit makes it quirky.

Cheers,
Stefan

martinwhitaker, 2018-04-30:
No, I think an import statement is allowed in the compilation unit scope. Follow through the equations in appendix A:
```
source_text
  description
    package_item
      package_or_generate_item_declaration
        data_declaration
          package_import_declaration
```
I did look at adding support for the missing items in the $unit scope, but the current design of the parser makes this hard work.

wallento, 2018-05-01:
Oh yes, you are right, sorry. Footnote 3 at `package_item` made me laugh :)

martinwhitaker, 2019-10-03:
Both the unit scope and imports into it are now supported in the master branch. The example provided by red0bear compiles without error.

closed 2019-10-03 by martinwhitaker

ppannuto, 2022-01-20:
Has this regressed? I ran into this issue with an example I was working on and found this thread. I tried [the example from red0bear](https://github.com/steveicarus/iverilog/issues/102#issuecomment-215404045) above, and it does not seem to work:
```
$ iverilog -g2012 fifo.sv fifo_parameters.sv
fifo.sv:2: syntax error
I give up.
```
I tried with both the current release
> Icarus Verilog version 11.0 (stable) ()

and a fresh compilation from the current git tip
> Icarus Verilog version 12.0 (devel) (s20150603-1214-g9898bffd)

Platform is OS X if that's relevant at all.

ppannuto, 2022-01-20:
Nope, I've just forgotten how these tools work. For any others coming across this in the future, file order is significant to icarus:

`iverilog -g2012 fifo_parameters.sv fifo.sv` 

works just fine.

---

Sorry for the noise.

martinwhitaker, 2022-01-20:
And for reference, section 26.3 of IEEE 1800-2012 states "The compilation of a package shall precede the compilation of scopes in which the package is imported.". The SystemVerilog syntax is ambiguous, and the parser needs to know whether an identifier is a package name to resolve some of the ambiguities.