steveicarus/iverilog #103: Internal error: Input vector

Kmanfi, 2016-04-28:
Hi,
I got this assertion with version 6e5ed73.
```
Internal error: Input vector expected width=7, got bit=1'bX, base=0, vwid=6
Assertion failed: (bits4_.size() == vwid), function filter_vec4, file vvp_net_sig.cc, line 900.
```

martinwhitaker, 2016-04-29:
Can you provide a small test case that exposes this problem. It will be difficult for us to identify the cause without that.

Kmanfi, 2016-04-29:
Code is about 1.4M lines of gate level verilog, so... not so easy to isolate the test case. Is there any other way I can help?

red0bear, 2016-04-29:
Could be a a netlist bad generated ? What program do you used to generate this netlist ?

Kmanfi, 2016-05-02:
I use Yosys. Actually I use Yosys' command test_autotb that uses iverilog. Test_autotb generate RTL, coarse net and gate net. Iverilog compiles RTL and coarse fine but gate level Verilog gives this assertion.
https://github.com/cliffordwolf/yosys

cliffordwolf, 2016-05-02:
@Kmanfi Can you put the file(s) you put into Yosys somewhere online, or send them to my gmail, so I can have a look at this?

Kmanfi, 2016-05-03:
Unfortunately I can't. Design has a lot of modules in hierarchy and I try to find which sub hierarchy or module make this assert. Any idea how to find this module in design haystack?

cliffordwolf, 2016-05-03:
@Kmanfi do you get the iverilog error during compile time (i.e. call to iverilog) or run time (i.e. call to vvp, or the generated executable)? If the latter, is it during simulation or during initialization of the design, i.e. do you need a test bench?

E.g. if you get the problem at runtime, but no test bench is required (bc. its during initialization), or in some cases if you get the problem at compile time, you can try choosing a different top module with the iverilog `-s <top_module>` option and narrow down the candidate module this way.

If you get the problem during compile time, you can try removing the module internals (everything except port declarations) in the iverilog verilog input one module at a time, starting at the modules that are at the leaves of the design hierarchy (i.e. modules that do not instantiate other modules) and work your way down from there.

Kmanfi, 2016-05-03:
@cliffordwolf I get assert during vpp. I'll try to narrow this with iverilog -s top module. Thank you for this.

martinwhitaker, 2016-05-03:
Can you get a gdb backtrace after the assertion failure, by
```
gdb vvp
run a.out
bt
```
(replace a.out with the actual compiler output file name).

This might give us enough information to deduce the source of the problem.

Kmanfi, 2016-05-04:
Here is attached bt.
[trace.txt](https://github.com/steveicarus/iverilog/files/248214/trace.txt)

martinwhitaker, 2016-05-08:
Although the trace provides a bit more information, I still can't reproduce the bug. So we really need a failing test case to get any further.

Kmanfi, 2016-05-09:
Sent isolated piece of code to @cliffordwolf .

cliffordwolf, 2016-05-09:
@Kmanfi: Got it. I'll try to create a minimal test case from that and then post my results here.

cliffordwolf, 2016-05-09:
Here is  a simplified test case:

```
module test;
  wire [5:0] foo;
  wire [6:0] bar;
  assign bar[5:0] = foo;
  assign foo[0] = 0;
endmodule
```

With current git head of icarus verilog (a219df2), I get the following error with that test case:

```
$ iverilog test.v && ./a.out 
Internal error: Input vector expected width=7, got bit=1'b0, base=0, vwid=6
vvp: vvp_net_sig.cc:900: virtual vvp_net_fil_t::prop_t vvp_wire_vec4::filter_vec4(const vvp_vector4_t&, vvp_vector4_t&, unsigned int, unsigned int): Assertion `bits4_.size() == vwid' failed.
Aborted
```

martinwhitaker, 2016-05-10:
@cliffordwolf : Thanks for that. I've reproduced and fixed the bug in both the master and v10 branches.

@Kmanfi : Can you check this fixes the bug in your full netlist, and close this issue if so.

@steveicarus : Note this fix broke the vvp_recv_vec4_pv test, which is actually trying to test the SV feature of allowing mixed continuous and non-blocking assignments to the same vector. I've fixed that by adding a delayed force operation to vvp, which I think is the proper way to handle this.

Kmanfi, 2016-05-11:
It works. Thank you.

closed 2016-05-11 by Kmanfi

martinwhitaker, 2016-05-11:
P.S. Using a delayed force to handle mixed continuous/non-blocking assignments is not a perfect fix - it will give wrong results if the user does an explicit force/release or tries to set the variable via the VPI - but it will serve until someone comes up with a better solution.