steveicarus/iverilog #105: Double tick and macro arguments do not work for macro names

udog14, 2016-06-06:
The example below defines **hello_world** macro by using another macro + argument.   The test  should print "Hello world".  Instead it reports that macro **hello_world** is undefined.

`define my_macro(a0) \
  `define hello_``a0 $display("Hello %s", `"a0`") 

module test();
  initial begin
    `my_macro(world);
    `hello_world;
  end
endmodule

I tested with the latest iverilog, version 11.0 (devel) (s20150603-368-g45fbf55)
Attached are two tests: 
- macro_arg1.v is the above.
- macro_arg2.v is slightly more complex using `ifdef.

[macro_arg.zip](https://github.com/steveicarus/iverilog/files/299749/macro_arg.zip)

Thank you!

martinwhitaker, 2016-06-06:
Agreed, this should work. I'll take a look at it.

martinwhitaker, 2016-06-06:
Fixed in both development and the v10 branch. Thanks for reporting this.

closed 2016-06-06 by martinwhitaker

udog14, 2016-06-06:
Martin, you are super amazing!  Thank you.