steveicarus/iverilog #106: VPI write during read-only phase is allowed

jeremyherbert, 2016-06-07:
As I discovered in https://github.com/potentialventures/cocotb/issues/448 , iverilog allows you to write to signals during the read-only phase over the VPI interface. The signal value persists at the same timestep if you read the signal back over VPI, but the vcd waveform is updated at the next simulated event, which confused me for quite a number of hours ;)

Is it possible to change iverilog so that writing during the read-only phase causes a warning or even an exception? This is a pretty subtle error to make, and I'd hate for someone else to waste as much time as I did on it.

jeremyherbert, 2016-06-07:
I'm also happy to give this a try myself if someone who understands the codebase could point me in the right direction.

jeremyherbert, 2016-06-07:
It seems that this error message is already present, but there seems to be some issue between the interaction of cocotb and iverilog. I'm going to close this until I work out what is really going on here.

jeremyherbert, 2016-06-07:
Reopening because the original issue is still valid. iverilog prints an error if you try to schedule a read/write trigger after a read-only trigger (which is the expected behaviour), but it still allows you to write to signals without error during the read-only phase.

martinwhitaker, 2016-06-09:
I've modified vpi_put_value to output an error message and squash the write if it is called during the read-only phase. Fix pushed to both the master and V10 branches. Thanks for reporting this.

closed 2016-06-09 by martinwhitaker