steveicarus/iverilog #107: No support to gate level zero time delay using setuphold/ifnone

red0bear, 2016-06-08:
Hello people, i was here trying to run a few files and gate level with time zero delay and i depared with theses messages:  
 
- Sorry:  ifnone with an edge-sensitive path is not supported.
  example : ifnone (AL => Z) = 0;

-  warning: timing checks are not supported and delayed signal
  example: $setuphold (posedge CLK, posedge ADDR, 0, 0, notifier,,, CLK_d, ADDR_d);

i would like to know if there are some effort to implement these features are they are just are in stand by.

martinwhitaker, 2016-06-09:
Speaking for myself, I have no plans to work on this. There's a lot that would need doing to get back-annotated simulation into a truly usable state.

red0bear, 2016-06-14:
hummmmm .... i see this could be good what you say. But i need to know if there is a list with features not implemented yet. Or at least what people are working on.

steveicarus, 2016-06-17:
The iverilog-devel mailing list is the place to go to ask
questions like this and get into the flow of things.

martinwhitaker, 2016-06-19:
Also, for standard Verilog, see the (admittedly incomplete) release notes for v10 on the Wiki for a list of major missing features, and also the Icarus Verilog bug tracker on SourceForge.

Unfortunately it's harder to find out what is/isn't supported for SystemVerilog.

red0bear, 2016-06-27:
i understand. But do you need help to handle this ? i would like to know what people are doing cause i feel bad coming here demanding something and you guys have a lot of work in stuff yet.

kdpatino, 2021-01-29:
Hi! Just want to ask if there is any update about this feature? :smile: I would like to move from VCS to iverilog but this is quite important.

caryr, 2021-01-29:
Short answer is no. We have been working on adding SystemVerilog and have not done any specify block improvements in a number of years.

kdpatino, 2021-01-29:
Ok! Thank You for your answer! :+1: