steveicarus/iverilog #109: Internal wires clash with user symbols

yugr, 2016-06-09:
The attached file currently causes trunk iverilog to return a rather mysterious error message:
```
iverilog bad.v
bad.v:8: error: Unable to bind wire/reg/memory `_s6' in `mod'
bad.v:8: error: Unable to elaborate r-value: _s6
2 error(s) during elaboration.
```
After some debugging I realized that _s6 in my code collides with internal auto-generated wire created by synthesizer. Not sure whether this can be fixed or not but I believe we can at least make the error more informative.

[bad.v.txt](https://github.com/steveicarus/iverilog/files/307661/bad.v.txt)

yugr, 2016-06-09:
Ok, here's a braindead patch which simply adds an assertion. Other alternatives I can think of
* use more unique template rather than _sN (_icarN?) - but this may hurt memory consumption
* keep increasing local counter in a loop until there are no clashes - sounds somewhat hacky
Should I pursue one of these?

[0001-Detect-name-clashes-with-user-symbols-early.patch.txt](https://github.com/steveicarus/iverilog/files/307677/0001-Detect-name-clashes-with-user-symbols-early.patch.txt)

martinwhitaker, 2016-06-09:
I fell foul of this one myself, and meant to get back to it. The comment for local_symbol() in netlist,h says "This method generates a non-hierarchical name that is guaranteed to be unique within this scope.", so I guess that's what it should do. The complication is that there is no single list of existing names; there's a map for signals, a map for parameters, etc. So checking for uniqueness could be quite time consuming,

I'll wait to see if Steve has anything to say on this one.

steveicarus, 2016-06-09:
Probably the simplest way to fix this is to change the local_symbol()
to return a symbol that starts with a "$" character. Currently, it
generates symbols starting with "_", but that is a perfectly legal
symbol for the user to enter, so I guess that won't work.

I'm reading the IEEE1364 standard, and it is  quite explicit that
starting with "_" is perfectly legal, but starting with "$" is
explicitly not, so there's that.

On 06/09/2016 02:10 PM, martinwhitaker wrote:
> I fell foul of this one myself, and meant to get back to it. The comment
> for local_symbol() in netlist,h says "This method generates a
> non-hierarchical name that is guaranteed to be unique within [...]

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

martinwhitaker, 2016-06-09:
That will need some fixing up in the vlog95 target. When synthesis is enabled, some local signals make it through to the output (that's where I fell foul of this). But that's a rare case, so we can probably afford to replace the $ with something longer that's unlikely to clash with a user symbol, maybe "_ivlsynth_".

martinwhitaker, 2016-06-09:
We might still have trouble with escaped identifiers.

steveicarus, 2016-06-09:
I expect it is possible for local symbols to be generated. The
vlog95 target should be able to handle this by escaping those
identifiers. The other targets, particularly vvp, should be fine
with this.

On 06/09/2016 02:38 PM, martinwhitaker wrote:
> That will need some fixing up in the vlog95 target. When synthesis is
> enabled, some local signals make it through to the output (that's where
> I fell foul of this). But that's a rare case, so we can [...]

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

steveicarus, 2016-06-09:
According to the standard, \foo is different from foo, so
$s61 should be different from \$s61. I don't remember if
Icarus Verilog handles this properly, but it should.

On 06/09/2016 02:46 PM, martinwhitaker wrote:
> We might still have trouble with escaped identifiers.

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

martinwhitaker, 2016-06-09:
??? My copy of the standard says the opposite!

steveicarus, 2016-06-09:
I stand corrected:

  3.7.1 Escaped Identifiers

  [...] Neither the leading backslash character nor the terminating
  white space is considerec to be part of the identifier. Therefore,
  an escaped identifier \cpu3 is treated the same as a nonescaped
  identifier cpu3.

So yeah, that part of the issue remains. But I still think that
the local symbols generator should generate symbol names that are
not legal in the source code (other then escaped) so that they do
not in any way pollute the name space.

yugr, 2016-06-10:
There another issue with temporary signals but this time more complicated. When exporting temp. signal to blif, module synthesizers (e.g. lpm_cmp_eq.cc) will typically use unchanged Verilog name for generated gates. So be it _sN, _$N, etc. - it'll be sent to ABC. Now when be try to export it back to Verilog from ABC, ABC will simply use the name it was provided with. So Verilog will always get it's own temporary names from ABC. This causes the testing pipeline iverilog -> ABC -> iverilog to break. I've just ran into this issue after switching to $-approach.

yugr, 2016-06-10:
What about dynamic selection of prefix? I.e. scan existing user names and then use _ssss (taking as many "s" as needed)?

yugr, 2016-06-10:
An example implementation (seems to fix issues with ABC). Sorry for messed formatting - I haven't found coding style description anywhere...

[0001-Avoid-name-clashes-with-user-symbols.-Fixes-GitHub-i.patch.txt](https://github.com/steveicarus/iverilog/files/308354/0001-Avoid-name-clashes-with-user-symbols.-Fixes-GitHub-i.patch.txt)

martinwhitaker, 2016-06-10:
> This causes the testing pipeline iverilog -> ABC -> iverilog to break.

Yes, this is the same  problem I had with the vlog95 target. I fixed it there by adding an extra leading "_" to local signals that were going to be exposed (done in the `exposenodes` functor).

As to your patch, I'd simply increment `lcounter_` until I found a unique identifier rather than extending the prefix. But as I mentioned above, it's an incomplete solution because it only checks for clashes with *signal* names.

steveicarus, 2016-06-10:
Hmm... This may work. Elaboration happens in phases, with
signals elaborated before expressions and gates, so user
symbols will (I think) be known before any local symbols
are created. I cringe at the possible performance implications
of this patch, though. Elaboration run times are already
not wonderful.

yugr, 2016-06-10:
I can measure timing. But as Martin pointed out this is really just a workaround, not a proper fix (as only local block symbols are tracked). I'm not really familiar with internals so I'd rather you and Martin decide what's the proper fix here.

martinwhitaker, 2016-06-19:
An alternative workaround for the round-trip issue (iverilog -> ABC -> iverilog) is to do what I did to solve the same issue with the vlog95 target, and add an extra leading "_" to all local symbols that find their way into the BLIF output. Of course this wouldn't be sufficient if there were more than two passes through iverilog in the pipeline.

steveicarus, 2019-09-25:
> An example implementation (seems to fix issues with ABC). Sorry for messed formatting - I haven't found coding style description anywhere... [...]

The problem is that when local_symbol() is called, not all user names are necessarily known. So if the collision comes after the _s6 is generated, then this won't fix anything.

I'm leaning towards just changing the prefix from _s to _ivl_. There are other cases where the _ivl_ prefix is used internally by Icarus Verilog.

steveicarus, 2019-09-25:
I've pushed an update that uses the \_ivl\_ prefix for local symbols, and does a better job assuring that the name is unique within the current scope. I think this is the best that can reasonably be done here.

closed 2019-09-25 by steveicarus