steveicarus/iverilog #117: VPI - delays ignored when registering callbacks

jgjverheij, 2016-08-02:
Hi,

I am trying to implement co-simulation between  [CλaSH](https://github.com/clash-lang/clash-compiler) and Icarus Verilog. 

By registering a callback, a pointer to an '_s_vpi_time_' structure must be given. For certain callbacks, Icarus Verilog _ignores_ the given time in the '_s_vpi_time_' structure. This is for example the case with the _cbReadWriteSynch_ and _cbReadOnlySynch_ callbacks.

I have attached a VPI example. In this example three callbacks are used, namely _cbAfterDelay_, _cbReadWriteSynch_, and _cbReadOnlySynch_. In the _cbAfterDelay_ handler, all three callbacks are registered (again) with a delay of 10, 1, and 9 respectively. 

By using ModelSim, the example outputs the following print-statements:
- After delay - current time 0
- Read write - current time 1
- Read only - current time 9
- After delay - current time 10
- Read write - current time 11
- Read only - current time 19
- After delay - current time 20
- Read write - current time 21
- Read only - current time 29
- After delay - current time 30
- Read write - current time 31
- Read only - current time 39
- After delay - current time 40
- Read write - current time 41
- Read only - current time 49
- Finish sim - current time 50

By using Icarus Verilog, the example outputs the following print-statements:

- After delay - current time 0
- Read write - current time 1
- Read only - current time 9
- After delay - current time 10
- Read write - current time 10
- Read only - current time 10
- After delay - current time 20
- Read write - current time 20
- Read only - current time 20
- After delay - current time 30
- Read write - current time 30
- Read only - current time 30
- After delay - current time 40
- Read write - current time 40
- Read only - current time 40
- Finish sim - current time 50

Only the first registration, of the _cbReadWriteSynch_ and _cbReadOnlySynch_ callbacks, has the correct timing (1 and 9). In the other executions the given time is ignored. 

Is this a bug or is this intended? If it is not a bug, why are the first executions at the correct time (1 and 9) and the others not?

[vpi.txt](https://github.com/steveicarus/iverilog/files/397035/vpi.txt)

martinwhitaker, 2016-08-07:
I was sure this had come up before, but I can't now find any discussion of it. The problem here is that if you read the IEEE standard, the wording used implies (without being absolutely clear) that the time specified for _cbReadWriteSynch_ and _cbReadOnlySynch_ callbacks is an absolute time. This is the behaviour that Icarus implements (with the maybe unexpected corollary that if the specified time is in the past, the callback is still executed, but in the current time slot). However, other simulators treat the specified time as being relative to the current time (and Stuart Sutherland's "The Verilog PLI Handbook" explicitly states this is the expected behaviour).

I would not be averse to changing Icarus to match the behaviour of other simulators, but Steve may think differently.

jgjverheij, 2016-08-08:
I am using Stuart Sutherland's book to create the VPI application and in there it is stated that the delay is defined relative to the current simulation time. You are right that the IEEE standard is not very clear about this. 

In which section of the IEEE standard do you read that the wording implies that the time is absolute? 
In section 27.33.2 of the IEEE 1364-2005, it is stated that the following situation will give an error:
"_Attempting to place a cbReadWriteSynch callback with a delay of **zero** at read-only synch time._"
This indicates that the given time is relative to the current simulation time, because it is not allowed to register a ReadWrite callback after a ReadOnly callback in the same simulation step.

I thought I solved it by only using the _ cbAfterDelay_ callback for delay statements and then registering the _cbReadWriteSynch_ and _cbReadOnlySynch_ with zero delay. I made the assumption that the delay was ignored and a zero delay does result in the current simulation step.

But if what you said is true and the delay is in absolute time, then the VPI application does not behave deterministically by using different simulators.

martinwhitaker, 2016-08-08:
> In which section of the IEEE standard do you read that the wording implies that the time is absolute?

In section 27.33.2, the description of _cbReadWriteSynch_ states "Callback shall occur after 
execution of events for a specified time". This is closer to the description of 
_cbAtStartOfSimTime_, "Callback shall occur before execution of events in a specified time
queue", than to the description of _cbAfterDelay_, "Callback shall occur after a specified amount of 
time".

> In section 27.33.2 of the IEEE 1364-2005, it is stated that the following situation will give an error:
> "_Attempting to place a cbReadWriteSynch callback with a delay of **zero** at read-only synch time._"
> This indicates that the given time is relative to the current simulation time...

I don't think this helps to clarify matters, as it is also stated that the following situation will 
give an error:

"Attempting to place a _cbAtStartOfSimTime_ callback with a delay of zero when simulation has
progressed into a time slice and the application is not currently within a _cbAtStartOfSimTime_
callback."

and I think all sources agree that _cbAtStartOfSimTime_ takes an absolute time.

jgjverheij, 2016-08-08:
I agree that _cbAtStartOfSimTime_ takes an absolute time. 

> I don't think this helps to clarify matters...

I meant this the other way around. In the same simulation time, a _cbReadWriteSynch_ cannot be scheduled after a  _cbReadOnlySynch_ callback. Why does the IEEE standard only state that with a delay of zero it will result in an exception? If it is absolute time for both callbacks, then the statement should have been: 
 "_Attempting to place a cbReadWriteSynch callback with the current simulation time as delay at read-only synch time._"

My initial question was: why does Icarus behave differently compared with ModelSim/QuestaSim when using the _cbReadWriteSynch_ and _cbReadOnlySynch_ callbacks? I think the answer is clear, ModelSim uses relative time and Icarus uses absolute time. 
Furthermore, the IEEE standard is not clear about this.

I do not know if this is problematic for MyHDL or Cocotb. But from my point of view, I would like to use the same VPI implementation for multiple simulators. It is of course possible to write simulator depended code, but that is not desirable. 

In the end it is yours (or Steves) decision how these features are implemented. I only wanted to show that the behavior of the callbacks are not the same compared with an other simulator.

steveicarus, 2016-08-08:
This would be a good conversation to have over on the iverilog-devel
mailing list, so as to hit a wider audience. This issue probably
matters to some people.

martinwhitaker, 2016-08-08:
> I meant this the other way around. In the same simulation time, a _cbReadWriteSynch_ cannot be
> scheduled after a  _cbReadOnlySynch_ callback. Why does the IEEE standard only state that with a
> delay [...]

I think you miss my point. You can't argue that this indicates that _cbReadWriteSynch_ takes a 
relative time without also arguing that the sentence I quoted indicates that _cbAtStartOfSimTime_ 
also takes a relative time. But let's just agree that the standard is badly written and move on. 
I'll put this out for discussion on the iverilog-devel mailing list in case anyone wants to argue 
for retaining the current behaviour in Icarus.

red0bear, 2016-09-18:
This topic the creator try run icarus with mixed API or program ?

jgjverheij, 2016-09-18:
In case of Verilog, I only used the Verilog Procedural Interface (VPI). The idea was to embed Verilog code in [CλaSH](http://www.clash-lang.org). The implementation connects (automatically) the 'lazy evalution' in CλaSH (Haskell) with the simulation steps (or clock-cycles) in Verilog. 

It is a bit comparable with Cocotb, in which co-simulation is defined between Python and Verilog.

red0bear, 2016-09-19:
Well, looks promising but i made something familiar with systemC. i see this like a particular change. But what the real effect to another's who use  VPI like me. But i didnt see any example showing the problem, i mean a .vcd show the wave difference. just that vpi.txt. Could you give a example using a simple IP or just how you want to mix the code "python/icarus"?

jgjverheij, 2016-09-20:
I did not use Python, but it is co-simulation between CλaSH (Haskell) and Verilog. The problem was (is) that delay statements are not the same when using different simulators. 

[image]

At the 'write' (w) moments, i write data into the Verilog simulator. At the end of a clock-cycle I read data from the simulator (the 'read' (r) moments). Writing happens with the _cbReadWriteSynch_ callback and reading happens with the _cbReadOnlySynch_ callback. 

The inputs and outputs are for example a list with values, in which every value correspond to a clock-cycle. This is thus an iterative process, which can be implemented in a loop (_for_, _while_) or using the Haskell functions _fold_ or _mapAccum_. The scheduling of the write and read moments repeats with a fixed delay (the length of the clock-cycle in this case). The main point of this topic is the question if these delays are _absolute_ or _relative_. As stated in the previous posts, Icarus Verilog uses an _absolute_ scheduling system and for example Modelsim uses _relative_ timings.

At the beginning of the co-simulation it is unknown for how many cycles the simulation has to run. So every cycle there is a check if there is a next cycle. If that is the case, callbacks for the next cycle are registered. 
For example, with the ModelSim the callbacks are registered with as timing the length of the clock-cycle, but in case of Icarus that does not work (because of the usage of absolute times instead of relative times). 

Is this description clear or do you want an actual example?

red0bear, 2016-09-21:
This last explanation is clear to first example you give when you opened the topic. But you could do 2 examples : 

1.VPI only "what you already made"
2.VPI with a simple verilog/systemverilog/vhdl/gatelevel "Choose what you like"  to you make your argumentation to change have more adepts. If you make a little vcd with times we could compare. 

Anyway if this is true against verilog, change sounds good. Thank you to take your time responding.

martinwhitaker, 2016-09-21:
No need for more examples as far as I am concerned - the issue is perfectly clear.

So far there has been no response to my post on iverilog-devel regarding this subject. I sent out a reminder last weekend, to give people one last chance to object to making the change.

martinwhitaker, 2016-09-25:
I've pushed a fix for this to the master branch. As it is a change in behaviour that isn't clearly a bug, I won't backport this to v10.

closed 2016-09-25 by martinwhitaker