steveicarus/iverilog #118: VHDL support is unable to handle Xilinx VHDL files

micrenda, 2016-08-03:
I have using iverilog to simulate a full TCP/IP implementation.
Unfortunately the only development board I have available is the Xilinx VC707. This means, to be able communicate via Ethernet, I am forced to use the Vivado IP Ethernet component (gig_ethernet_pcs_pma).

Vivado IP component ara composed by a big section in Verilog (which seems to be supported by iverilog) and a small file in VHDL.

The problem is that the VHDL support of iverilog seems to be incomplete and I get this message:
```
gig_ethernet_pcs_pma_v15_1_1/hdl/gig_ethernet_pcs_pma_v15_1_rfs.vhd:193: syntax error
gig_ethernet_pcs_pma_0/gig_ethernet_pcs_pma_v15_1_1/hdl/gig_ethernet_pcs_pma_v15_1_rfs.vhd:193: error: Syntax error in block declarations.
```
It seems it is related to the missing support of the attribute section:
```
  -- These attributes will stop timing errors being reported in back annotated
  -- SDF simulation.
  attribute ASYNC_REG                       : string;
  attribute ASYNC_REG of data_sync_reg1    : label is "true";
  attribute ASYNC_REG of data_sync_reg2    : label is "true";
  attribute ASYNC_REG of data_sync_reg3    : label is "true";
  attribute ASYNC_REG of data_sync_reg4    : label is "true";
  attribute ASYNC_REG of data_sync_reg5    : label is "true";
  attribute ASYNC_REG of data_sync_reg6    : label is "true";
```
If you want, I can send a full reproducible example privately (just contact me via github).

Thank you

martinwhitaker, 2016-08-08:
As a workaround, does commenting out the attribute statements enable you to simulate this component?

Icarus does not support these attributes anyway, so they wouldn't have any effect even if the VHDL preprocessor accepted them.

steveicarus, 2022-05-15:
Cleanup. This issue seems to be stale. Martin received no response to his query, and VHDL is not a high priority anyhow. If there is interest, please create a new issue.

closed 2022-05-15 by steveicarus