steveicarus/iverilog #121: executable not found

red0bear, 2016-08-23:
i just update now and compiled and when i try execute this appear to me : 

sh: /usr/local/lib/ivl/ivlpp: File or directorie not found
sh: /usr/local/lib/ivl/ivl: File or directorie not found

steveicarus, 2016-08-23:
Did you do the "install" step?

On 08/23/2016 09:32 AM, Felipe Fernandes da Costa wrote:
> i just update now and compiled and when i try execute this appear to me :
> 
> sh: /usr/local/lib/ivl/ivlpp: File or directorie not found
> sh: /usr/local/lib/ivl/ivl: File or directorie not found

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

red0bear, 2016-08-23:
Yes , i did and tried call a execution .... but give only this :(

red0bear, 2016-08-23:
if this help : 

Compilation output: 

/usr/bin/install -c ./ivlpp "/usr/local/lib64/ivl/ivlpp"

./../mkinstalldirs "/usr/local/lib64/ivl"

steveicarus, 2016-08-23:
Looks like a configuration mismatch between parts of the
tool chain. Are you installing from git? If so, try:

make clean
sh autoconf.sh   # Make sure the configure script is up to date
./configure
make all
make install

red0bear, 2016-08-23:
Still giving the error after a i made this what you commented....... i checked dirs here and see : 

/usr/local/lib/ivl/ivlpp  -->  this is what program want 
/usr/local/lib64/ivl/ivlpp --> where is really installed 

/usr/local/lib/ivl/ivl  --> this is what program want 
/usr/local/lib64/ivl/ivlpp  --> where is really installed

steveicarus, 2016-08-23:
Check the (prefix) and (exec_prefix) values in all the Makefiles
that were generated by configure, and also check the date on the
"iverilog" binary. The Makefile compiles the IVL_ROOT value into
the iverilog binary, and it should get that value from the Makefiles.
It looks like the iverilog binary is getting an old value, so either
you have mixed up Makefiles, or the iverilog binary needs to be
rebuilt to get the proper values from the Makefile.

red0bear, 2016-08-23:
I made a new git clone and i get again the error , at least i resolved it by making a linking where iverilog wants to call 
ln -s /usr/local/lib64/ivl/ /usr/local/lib/ivl

martinwhitaker, 2016-08-23:
This is fallout from pull request #116. I'll try to fix it.,

martinwhitaker, 2016-08-23:
Should be fixed now. Not yet tested on Windows.

red0bear, 2016-08-23:
Well to linux after i deleted link it worked here after i recompiled all thing. :Sorry im not Rwindows user.

martinwhitaker, 2016-10-15:
Tested on Windows.

closed 2016-10-15 by martinwhitaker