steveicarus/iverilog #124: input buffer overflow

Kmanfi, 2016-09-07:
Please run attached verilog file.
File is auto generated by Yosys. @cliffordwolf 

iverilog reject_bug.v 
`input buffer overflow, can't enlarge buffer because scanner uses REJECT`
[reject_bug.v.zip](https://github.com/steveicarus/iverilog/files/459027/reject_bug.v.zip)

martinwhitaker, 2016-09-08:
This is a limitation of the scanner generated by flex, caused by you having a single lexical item (a string) that is bigger than the scanner input buffer size. You can work round this by adding "-DYY_BUF_SIZE=32768" to CPPFLAGS in the top level Makefile.

Looking at the lexor.lex rules, I'm not convinced we need to use REJECT. The first place it is used, disabling the detection of decimal numbers when in UDPTABLE state, could be handled by making that rule conditional on being in INITIAL or EDGES state. The other places it is used are to prevent recognition of Verilog-AMS or SystemVerilog syntax, but I think all those could be replaced by an explicit error. I don't think there is any valid "traditional Verilog" syntax that would be masked by doing this (in the case of the SystemVerilog scaled time values, almost certainly not, as the test for a SystemVerilog generation is broken).

steveicarus, 2016-09-08:
And eliminating the use of REJECT would help how? I agree that
there are a few cases where REJECT are not needed, but most make
the lexor (and resulting parser) cleaner. Do and of the uses of
REJECT impact the reported issue?

On 09/08/2016 11:44 AM, martinwhitaker wrote:
> This is a limitation of the scanner generated by flex, caused by you
> having a single lexical item (a string) that is bigger than the scanner
> input buffer size. You can work round this by adding [...]

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

steveicarus, 2016-09-08:
OK, now I understand how removing REJECT use would help.
Still, it looks like it may not be so obviously easy.

On 09/08/2016 12:26 PM, Stephen Williams wrote:
> And eliminating the use of REJECT would help how? I agree that
> there are a few cases where REJECT are not needed, but most make
> the lexor (and resulting parser) cleaner. Do and of the uses of
> REJECT [...]

-- 
Steve Williams                "The woods are lovely, dark and deep.
steve at icarus.com           But I have promises to keep,
http://www.icarus.com         and lines to code before I sleep,
http://www.picturel.com       And lines to code before I sleep."

martinwhitaker, 2016-09-08:
Seems fairly straightforward to me. Can you see anything wrong with this?

[lexor.patch.txt](https://github.com/steveicarus/iverilog/files/462831/lexor.patch.txt)

It doesn't break anything in the test suite (even after modifying the udp_jkff test to have no spaces between the j and k columns), and I've checked the new error messages appear when expected.

steveicarus, 2016-09-08:
Superficially, it looks fine, and may in fact be perfectly
correct. Question: what happens to this string before and
after your change:

   9.5m

Before, this becomes two tokens, a string and an identifier,
after it becomes an error unless AMS is enabled. Perhaps that
is OK?

martinwhitaker, 2016-09-08:
Hmm, no. I thought there was a requirement for white space to separate numbers and identifiers, but careful reading of the standard suggests not. So looks like we are stuck with a fixed input buffer size.

Kmanfi, 2016-09-09:
Changed 
`CPPFLAGS = -DHAVE_CONFIG_H $(INCLUDE_PATH)`
to
`CPPFLAGS = -DHAVE_CONFIG_H $(INCLUDE_PATH) -DYY_BUF_SIZE=32768`
Compiled Master HEAD

iverilog reject_bug.v 
```
reject_bug.v:9: syntax error
I give up.
```

martinwhitaker, 2016-09-09:
That's correct. You have a `begin` without a corresponding `end`.

Kmanfi, 2016-09-09:
Oops. My mistake.
Now it works.
Thank you.

closed 2016-09-09 by Kmanfi