steveicarus/iverilog #126: undefined reference to `lexor_keyword_mask'

jiapei100, 2016-09-13:
Current GIT brought me the following error messages:

> main.o: In function `main':
> /home/jiapei/Downloads/hardware/iverilog/main.cc:939: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/main.cc:954: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/main.cc:958: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/main.cc:961: undefined reference to `lexor_keyword_mask'
> lexor.o: In function `VLlex()':
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:684: undefined reference to `lexor_keyword_mask'
> lexor.o:/usr/include/c++/5/ext/new_allocator.h:104: more undefined references to `lexor_keyword_mask' follow
> lexor.o: In function `VLlex()':
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:280: undefined reference to `lexor_keyword_code(char const*, unsigned int)'
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:633: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:636: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:641: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:647: undefined reference to `lexor_keyword_mask'
> /home/jiapei/Downloads/hardware/iverilog/lexor.lex:654: undefined reference to `lexor_keyword_mask'
> lexor.o:/home/jiapei/Downloads/hardware/iverilog/lexor.lex:662: more undefined references to `lexor_keyword_mask' follow
> collect2: error: ld returned 1 exit status
> Makefile:222: recipe for target 'ivl' failed
> make: *** [ivl] Error 1
> 

Cheers
Pei

martinwhitaker, 2016-09-13:
I can't reproduce this with a clean checkout and build:
~~~
git clone https://github.com/steveicarus/iverilog.git
cd iverilog
sh autoconf.sh
./configure
make
~~~
What O/S are you using, and what compiler? What version of gperf? When you ran make, did the call to gperf that generates the lexor_keyword.cc file run without any errors. If so, what is the output from
~~~
nm lexor_keyword.o
~~~

steveicarus, 2016-09-14:
I tried to replicate it as well, and no luck. I think that Martin is right,
something
broke with your lexor_keyword.cc file. The "sh autoconf.sh" step makes that
file using gperf. You sure you have gperf installed?

martinwhitaker, 2016-11-25:
As there's been no response for over 2 months, I'm going to assume it was a local problem and is now fixed. Please feel free to reopen if there is still an issue.

closed 2016-11-25 by martinwhitaker

brooksbp, 2017-06-04:
I just hit this issue.  Installing gperf resolved it.