steveicarus/iverilog #127: Simulation when port width mismatch

nadezhin, 2016-09-17:
There is yet another open source Verilog compiler. It is part of ACL2 proof checker:
http://www.cs.utexas.edu/users/moore/acl2/manuals/current/manual/?topic=ACL2____VL
It contains some tests where its behaviour is compared with commerical NCV and VCS Verilog complilers. Recently we compared its behaviour also with the Icarus Verilog. We want to report found differences in case they might be useful.

The first issue was found when port instance is connected with mismatched width. Icarus Verilog prints warnings and continues simulation. It's good but simulation results differ from NCV/VCS. Perhaps Icarus could mimic their results.
To reproduce this:
1) Download two files into directory ${COSIMS}:
https://github.com/acl2/acl2/blob/master/books/centaur/sv/cosims/test.sv
https://github.com/acl2/acl2/blob/master/books/centaur/sv/cosims/fourvalued.data
and the third file into directory ${COSIMS}/inst_size3:
https://github.com/acl2/acl2/blob/master/books/centaur/sv/cosims/inst_size3/spec.sv
2)

    cd ${COSIMS}/inst_size3
    iverilog -g2012 spec.sv ../test.sv -Dinfile=\"../fourvalued.data\" -Doutfile=\"outputs.iv.data\" && ./a.out
3) The file ${COSIMS}/inst_size3/outputs.iv.data starts from lines

    // 0x00000000
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzzzzzzz00zzzzzzz00zzzz00zzz00zz00z00000
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzzzzzzz01zzzzzzz01zzzz01zzz01zz01z01011
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzzzzzzz10zzzzzzz10zzzz10zzz10zz10z10100
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzzzzzzz00zzzzzzz00zzzz00zzz00zz00z00000
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzzzzzzz00zzzzzzz00zzzz00zzz00zz00z00000

though the expected output is 

    0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzz00zzzz00zzz00zz00z0000x
    0000000000000000000000000000000000000000000000000000000000000000000000000000000000000001010101010101zzzzz01zzzz01zzz01zz01z0101x
    0000000000000000000000000000000000000000000000000000000000000000000000000000000000000010101010101000zzzzz10zzzz10zzz10zz10z1010x
    0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzz00zzzz00zzz00zz00z0000x
    0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000zzzzz00zzzz00zzz00zz00z0000x

There are other tests in the repository rejected by Icarus Verilog.
They are marked by empty marker files

    ${COSIMS}/*/no_icv

martinwhitaker, 2016-09-17:
Most of the differences are due to the combination of mismatched widths in the port connections and the fact that you have deliberately mislabelled the port directions. Icarus does not properly handle the coercion of an output to inout. I'll push a fix for this shortly.

martinwhitaker, 2016-09-17:
I've pushed a fix to both the master and v10 branches to make the compiler correctly handle output port coercion. I don't agree with your expected results for the case where the connected signal is narrower than the port width. If you disagree, please take a look at the simplified tests:

  https://github.com/steveicarus/ivtest/blob/master/ivltests/br_gh127c.v
  https://github.com/steveicarus/ivtest/blob/master/ivltests/br_gh127f.v

and explain where and why they should be different.

Thanks for reporting this.

nadezhin, 2016-09-18:
Thank you for a quick fix.
I ran br_gh127?.v files on the ncverilog compiler (version 10.20-s040).
Tests b,c,e,f PASSED , but tests a,d FAILED.

The listing of br_gh127a.v ends with
ncsim> run
00 : 0 z0 : 00 00 : 000 00
01 : 1 z1 : 01 01 : 001 01
10 : 0 z0 : 10 10 : 010 10
11 : 1 z1 : 11 11 : 011 11
FAILED

The listing of br_gh127d.v ends with
ncsim> run
00 : 00 0 : 00 00 : 00 z00
01 : 01 1 : 01 01 : 01 z01
10 : 10 0 : 10 10 : 10 z10
11 : 11 1 : 11 11 : 11 z11
FAILED

I will try to read carefully the Verilog standard to understand whether it
is correct behavior.
It will take some time.

martinwhitaker, 2016-09-18:
The standard doesn't give explicit rules for mismatched port widths, so is open to interpretation (that's why Icarus outputs warning messages, telling you what it is going to do). I think the key paragraph in the standard is:

> Each port connection shall be a continuous assignment of source to sink, where one connected item shall be a signal source and the other shall be a signal sink. The assignment shall be a continuous assignment from source to sink for input or output ports. The assignment is a non-strength-reducing transistor connection for inout ports.

(from IEEE 1800-2012 section 23.3.3).

br_gh127{a,d}.v test the behaviour for input and output ports.. You would expect a continuous assignment to pad with zeros when assigning from a smaller to a larger width and to truncate when assigning from a larger to a smaller width.

br_gh127{b,c,e,f}.v test the behaviour for inout ports (both explicit and coerced). There are no semantics for connecting different width nets with tran gates, so Icarus chooses to connect as many bits as it can, and leave the excess bits dangling.

nadezhin, 2016-09-18:
I looked at the standard and I agree that Icarus Verilog results in
br_gh127{a,d}.v match the standard better than  Ncverilog results.

Also I ran Ncverilog on the initial test inst_size3. Its output matches
exactly the output of updated Icarus Verilog.
Output of ACL2 Verilog differs from them and I consider that this is ACL2
bug.

So I am satisfied with this fix. Thank you for it. The issue can be closed.

martinwhitaker, 2016-09-20:
Thanks. Closing.

closed 2016-09-20 by martinwhitaker