steveicarus/iverilog #128: "Unsupported LPM type: 36" error when converting to VHDL

udif, 2016-10-27:
A friend off mine encountered the error above when trying to convert his verilog code to VHDL using Icarus.
Unfortunately, the source code causing the offending error cannot be included here (I don't have any access to it either).
I think he has used the latest Windows download (10.1.1) available at http://bleyer.org/icarus/
I tried searching for this error in the icarus source code and I see it involves IVL_LPM_CONCATZ which is not handled in the VHDL target, but I couldn't find what is causing this exactly, as I'm not familiar with Icarus' code base.

```
C:\Users\xxx\Documents\Verilog>iverilog -o ddfe_TB -t VHD -c comp_list.txt
ERROR: Unable to read config file: C:\iverilog\lib\ivl\VHD.conf
error:  failed to load.
: The specified module could not be found.

error: Code generator failure: -2

C:\Users\xxx\Documents\Verilog>iverilog -o ddfe_TB -t vhdl -c comp_list.txt
VHDL conversion error: Unsupported LPM type: 36
VHDL conversion error: Failed to translate LPM dfe50g_15taps_top._s636
VHDL conversion error: Unsupported LPM type: 36
VHDL conversion error: Failed to translate LPM dfe50g_15taps_top._s639
VHDL conversion error: Unsupported LPM type: 36
VHDL conversion error: Failed to translate LPM dfe50g_15taps_top._s642
error: Code generation had 6 error(s).

```
thanks,
Udi

klammerj, 2016-10-29:
They have:
      IVL_LPM_CONCATZ = 36, /* Transparent concat */
in ivl_target.h
(I think it passes through Z values or sthg...)

klammerj, 2016-10-29:
You can try the -N output.net option and look at the resulting file...
I think it gets written even if the output module fails...

martinwhitaker, 2016-10-29:
A `CONCATZ` is generated when several contiguous parts of a net are driven by different continuous assignments, e.g.
~~~
wire [1:0] a;
wire [2:0] b;
wire [2:0] c;
wire [7:0] d;

assign d[1:0] = a;
assign d[4:2] = b;
assign d[7:5] = c;
~~~
The compiler optimises this into a single concatenation (equivalent to `d = {c, b, a}`, but supporting different drive strengths on each part).

There is a configuration option that disables this optimisation. In the file `/usr/local/lib/ivl/vhdl.conf` add the line
~~~
flag:DISABLE_CONCATZ_GENERATION=true
~~~
We should probably make this the default for the vhdl target; it's already used for the vlog95 target.

martinwhitaker, 2016-11-25:
I've disabled CONCATZ generation for the VHDL target, which should solve this issue.

closed 2016-11-25 by martinwhitaker