steveicarus/iverilog #129: Nested for loops with SystemVerilog loop variables crashes compiler

plotkin1996, 2016-11-23:
To reproduce:
```verilog
module test;
  initial
    for(int i=0;i<10;i++)
      for(int j=0;j<10;j++)
        #10 $display(i,j);
endmodule
```
The compiler hangs up for a while and then crashes:
```
sh: line 1: 20533 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg23001898e" -f"/tmp/ivrlg3001898e" -p"/tmp/ivrli3001898e"
     20534 Segmentation fault      (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh3001898e" -C"/usr/lib64/ivl/vvp.conf" -- -
```

martinwhitaker, 2016-11-24:
You don't say what version you're using, but I assume v10, as I can reproduce this bug with the head of the v10 branch, but not with the head of the master branch. I'll try to find and backport the change that fixed this.

martinwhitaker, 2016-11-25:
I've applied the fix for this to the v10 branch.

closed 2016-11-25 by martinwhitaker