steveicarus/iverilog #131: Force/release simulation issue with flash model

olofk, 2016-11-28:
I'm trying to simulate a Micron Flash memory model (https://www.micron.com/~/media/documents/products/sim-model/nor-flash/serial/bfm/n25q/n25q128a_micronxip_hold_18v_vg11,-d-,tar.gz). It works fine in ModelSim and Riviera Pro, but fails in Icarus (tried latest git head), as I only get 'x' on the MISO line (DQ1)

It seems like memory model writers always use the most arcane features of verilog, and this issue seems to be related to a force/release of a wire. Commenting line 2024 (`force DQ1 = 1'bX;`) in `N25Q128A11E_VG12/code/N25Qxxx.v` makes it work. Haven't dug deep enough into the LRM to figure out if this is a issue with Icarus or if the model should be changed.

martinwhitaker, 2016-11-28:
It's a race in the Micron model. Removing unrelated statements, they have written:
~~~
fork : CP_sendToBus
  force DQ1 = 1'bX;
  force DQ1 = bitOut;
join
~~~
The two `force` statements are therefore notionally executed in parallel, and the simulator is free to execute them in either order.

Looking at the rest of that block, I suspect they meant to write begin/end, not fork/join.

martinwhitaker, 2017-01-29:
As this appears to be a bug in the Micron code, not in the compiler, closing. Please reopen if you disagree.

closed 2017-01-29 by martinwhitaker