steveicarus/iverilog #137: assign using posedge clock freeze simulation

red0bear, 2016-12-11:
Merry Xmas to all. 

Well im trying to make a counter from recovery clock to a space wire and run on icarus verilog. Bellow follow the code where i want make a counter using a assign:

`wire [4:0] counter;`
`wire  [4:0] counter_pos;`
`wire  [4:0] counter_neg;`
`wire posedge_clk;`
`wire negedge_clk;`	

`assign  counter_pos = (!rx_resetn)?5'd0:(counter == 5'd3 & control[2:2])?5'd0:(counter == 5'd9)?5'd0:(posedge_clk)?counter_pos+5'd1:counter_pos;`
`assign counter_neg = (!rx_resetn)?5'd0:(counter == 5'd3 & control[2:2])?5'd0:(counter == 5'd9)?5'd0:(posedge_clk)?counter_neg+5'd1:counter_neg;`

Clock recovery is given bye IEEE 1355-1995 and i get last week trying to understand why icarus verilog freeze with this code. I have been tried use registers instead assign and it works fine but don't give the result where i expect from assign. 

i will post more information more late.

klammerj, 2016-12-11:
This might be because there's counter_neg appearing on the right side and the thing is asynchronous. 
I've seen the simulator go into endless loops on combinatorial cycles before. 
I wonder what this line is actually supposed to do...

steveicarus, 2016-12-11:
Sure looks to me like a combinational cycle. This is the logi design equivalent of an infinite loop.

red0bear, 2016-12-11:
Thank you to respond. 

The initial idea i wrote is in line  96 - 101. This didn't work and freeze simulation. After this i tried use register instead assign. They are on line 149 - 195 , and it works fine and don't freeze.   You can test just changing line 59 and 60 to reg and comment counter_pos and counter_neg assign. This logic is to be used to recovery a clock from Data and Strobe from SpaceWire. Data come coded in posedge and negedge clock.

red0bear, 2016-12-13:
Well , i remake to rtl avoiding use from assign but using the same idea. Now it  don't freeze. i updated on git a new version.

martinwhitaker, 2017-01-29:
This appears to have been an error in the user's code, not in the compiler, so closing.

closed 2017-01-29 by martinwhitaker