steveicarus/iverilog #138: Verilog using -D problems

red0bear, 2016-12-11:
Verilog using -D with vpi don't see signals when you make a call . im trying to make configuration to use to different verilog sources top instance. But when i call a vpi it fail to find signal on top instance. 

example using -D :
iverilog -oenv_global_spw.vvp -D VERILOG_LIGHT ../rtl/RTL_VB/*.v ../testbench/module_tb.v

caryr, 2017-12-17:
The -D flag is used to control how things are compiled. The VPI interface is part of the run time. You will need to provide more details regarding your issue because what you have provided here is not enough. Note that plus arguments can be used to change run time behaviour.

caryr, 2017-12-18:
Can this one be closed as well or if not can you give a better example of what you are trying to do so we can help?

red0bear, 2017-12-18:
Yes, sure.

caryr, 2017-12-19:
Okay, thanks.

closed 2017-12-19 by caryr