steveicarus/iverilog #139: Icarus treats all unqouted parameters containing 'e' as a float

olofk, 2016-12-16:
Using a top-level looking like this

    module m #(parameter p = 0);
      initial $display(p);
    endmodule

I noticed that passing a parameter -Pm.p=true evaluates to 0.000. I guess the correct solution is to pass -Pm.p="true" instead, but I got a bit surprised when I noticed that every unqouted string containing the letter 'e' was parsed as a float. I would argue for either a stricter test for floats and/or treating it as a string if it fails to be parsed as a int or float

Also, probably a non-argument, but ModelSim, Isim and Xsim all treat the unquoted string as a string, that evaluates to 0x74727565

olofk, 2016-12-17:
Some additional tests revealed that passing 0x1d triggers an assertion in isdigit, vhile 0x1e is treated as a float with value 30.0000

martinwhitaker, 2017-01-29:
Thanks for reporting this. I've added much better error checking for defparam values supplied on the command line, which should have fixed all the issues you reported. Pushed to both the master and v10 branches.

Note that values should be specified using Verilog syntax; C syntax (e.g.0x1d) is not supported.

closed 2017-01-29 by martinwhitaker