steveicarus/iverilog #140: PLI warnings and seg fault

markNZed, 2016-12-21:
I would like to use a VCD reader from http://chris.spear.net/pli/ designed for VCS with Icarus. I've tried to modify it and run with Master branch 063ae775681

I see warnings like:

WARNING, unregonized $var type 'ire', using 'wire'
WARNING, unregonized $var type 'eg', using 'wire'

But the Verilog looks correct.

It then fails with a seg fault. The attached taf.gz should have everything to reproduce, just run make.

[vcdr.tar.gz](https://github.com/steveicarus/iverilog/files/666670/vcdr.tar.gz)

markNZed, 2016-12-21:
The WARNING was output by the VCD Reader and I've fixed that. But the seg fault is still there. Are there any guidelines on how to debug a seg fault with Icarus ?

markNZed, 2016-12-21:
The verbose mode in the VCD reader causes the seg fault so I can work around that for now. The next issue is he VCD reader uses tf_strdelputp which is not defined by Icarus:

vvp: symbol lookup error: ./vcdr.vpi: undefined symbol: tf_istrdelputp

Any suggestions much appreciated.

martinwhitaker, 2016-12-21:
The TF routines were deprecated (and no longer documented) in the 2005 standard, so not a lot of effort has been put into implementing them. What there is, you'll find in the libveriuser sub-directory of the Icarus source code. I've never looked at any of this stuff myself, so can't say how easy or hard it would be to add the missing routines.

markNZed, 2016-12-21:
Thanks Martin. I'm wondering if I can replace the tf_strdelputp with a VPI call. Basically we need a delayed assignment using a string value. Is it possible to mix the TF and VPI ?

markNZed, 2016-12-22:
I used vpi_put_value with a delay to replace tf_strdelputp and ran into the next issue. From inside a tf_misc function it calls tf_setrealdelay then there is a seg fault in veriusertfs.c:305 I guess this is an issue in Icarus related to how the user_data is managed,  the seg fault is probably related to assigning pli in the previous line. 

pli = (p_pli_data)data->user_data;
tf = pli->tf;

closed 2016-12-22 by markNZed