steveicarus/iverilog #141: $display causing seg fault with PLI

markNZed, 2016-12-22:
I'm driving signals through the VPI e.g.

vpi_put_value(wr_handle, &val, &delay, vpiInertialDelay || vpiForceFlag);

and in the Verilog testbench it runs fine if I don't use $display. I can use $display before initialising the VPI/PLI code. But if I use $display during the simulation I get a seg fault.

```
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f519b5b0428 in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007f519b5b0428 in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007f519b5b202a in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007f519b5a8bd7 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007f519b5a8c82 in __assert_fail () from /lib/x86_64-linux-gnu/libc.so.6
#4  0x00000000004b93fb in vpi_put_value (obj=<optimized out>, vp=vp@entry=0x7ffc1c6a8790, when=when@entry=0x7ffc1c6a87a0, flags=flags@entry=1) at vpi_priv.cc:1062
#5  0x00007f519b367280 in vcdread_changes () at vcdr_pli.c:362
#6  0x00007f519b367707 in vcdr_misc (data=<optimized out>, reason=10, paramvc=<optimized out>) at vcdr_pli.c:564
#7  0x00000000004b3135 in sync_cb::run_run (this=0x1d0e490) at vpi_callback.cc:319
#8  0x000000000048ab4e in generic_event_s::run_run (this=0x71d510 <generic_event_heap+1456>) at schedule.cc:502
#9  0x000000000048cf68 in schedule_simulate () at schedule.cc:1183
#10 0x0000000000444584 in main (argc=<optimized out>, argv=0x7ffc1c6a9b38) at main.cc:501
 
```

Apart from this I have the PLI based VCD replay from http://chris.spear.net/pli/ working in Icarus. I'd like to resolve the $display issue then Chris could make the package available.

I've attached the demo. To see it running without the seg fault comment out `$display("hello"); `in tb_read.v Any hints much appreciated!

[vcdr2.tar.gz](https://github.com/steveicarus/iverilog/files/669956/vcdr2.tar.gz)

markNZed, 2016-12-23:
It seems a call to $display causes acc_handle_tfarg to fail when used in a misctf callback.

klammerj, 2016-12-24:
the failing line is:  assert(obj);
what are you passing as obj to vpi_put_value()?

> It seems a call to $display causes acc_handle_tfarg to fail when used in a misctf callback. 

??? I'm not seeing any of those in the call-trace you posted.

markNZed, 2016-12-24:
Thanks for taking a look. The code in vcdr_pli.c looks like:     
```
vpiHandle wr_handle;
wr_handle = acc_handle_tfarg(signal->argn);
vpi_put_value(wr_handle, &val, &delay, vpiInertialDelay || vpiForceFlag);
```

wr_handle is passed in as obj.

klammerj, 2016-12-24:
there's a:
>       if (n > 0) {
[...]
>       } else {
>             rtn_h = (vpiHandle) 0;
>       }
> 
in acc_handle_tfarg()
(file a_handle_tfarg.c)

The n is the argument to the function. 
What n was passed? That or it didn't find the arg...

PS: you can download the verilog source code from github. 
PPS: It's actually an abort, not a SEGFAULT.

markNZed, 2016-12-24:
You're right it is an abort. The n passed is correct for the calltf function (e.g. argument 6 from 18) associated with this misctf call. It works fine if we don't use the $display. The thing that makes me think this is an issue with Icarus and not the PLI code is the ability to comment in/out $display and cause/avoid the abort. I guess Icarus should guarantee that a $display call does not change the behavior of other PLI code.

klammerj, 2016-12-25:
any Idea what those messages mean?
or where they come from?
> Vcdr_Sigp[VRSP] $vcdread_signal_pairs arguments 7 & 8 are not register, string pair
> VCDR[YECH] String was 'test.d'
> 
(right at the start of the vvp run)
might be significant, given that it fails when reading the test.d thingy...

markNZed, 2016-12-25:
That is my mistake in tb_read.v when two Verilog signals are registered to the same VCD signal name. I've cleaned that up but it does not resolve the issue. In the attached I've commented out the $display call, so you should see "Summary: There were 0 mismatches." which indicates the test ran OK. Then comment in line 63 `$display("hello");` and we get the abort.

The call $vcdread_signal_pairs tells the PLI which Verilog signals should match up with which names in the VCD file. The function vcdr_sigp_call implements $vcdread_signal_pairs in vcdr_pli.c and at the end of that function it calls tf_synchronize which schedules the misctf function vcdr_misc. From vcdr_misc it calls vcdread_changes to read the VCD and schedule changes to the signals using vpi_put_value, then vcdr_misc calls tf_setrealdelay to schedule a callback for the next simulation time marked in the VCD file.

It seems that a call to $display is somehow corrupting the context for vcdr_misc so the argument number used by acc_handle_tfarg in vcdread_changes is no longer matching with the arguments that were provided to $vcdread_signal_pairs. 
[vcdr2.tar.gz](https://github.com/steveicarus/iverilog/files/672119/vcdr2.tar.gz)

klammerj, 2016-12-25:
ah, ok...
you're correct
stepping into the iterator I'm getting:
> (gdb) step
> 55            res->nargs = nargs;
> (gdb) p nargs
> $54 = 1
> 
which kinda suggests that it is the argument list of the $display
and printing the element gives:

> 
> (gdb) p (char *)args[0]
> $56 = 0x9124370 "Xu\025\b\250A\022\t\005"
> (gdb) step
> 
(which does not really tell me anything...)
and I can't seem to figure out where this gets set... there's some inheritance going on...
Maybe they don't actually store any of the contexts?

markNZed, 2016-12-26:
We are making some progress, thanks. When vcdr_misc calls tf_setrealdelay this means iverilog/libveriuser/veriusertfs.c line 417 establishes the callback with `vpi_register_cb(&cb)` and line 400 `p_pli_data pli = vpi_get_userdata(sys);` establishes the context (it seems the PLI context is stored in the user_data of the callback). 

My best guess is that line 428 in iverilog/libveriuser/veriusertfs.c is returning the $display context when calling tf_getinstance() Is this something you could check ?

klammerj, 2016-12-27:
Here's a patch. 
It seems to fix the bug. 
Hope it applies cleanly(haven't pulled in a while). 

[0001-fix-failing-VPI-callbacks.patch.txt](https://github.com/steveicarus/iverilog/files/674186/0001-fix-failing-VPI-callbacks.patch.txt)

markNZed, 2017-01-01:
@klammerj sorry for the delayed reply. You are a genius, thank you! I ran the patch on branch v10_1 and experimented with various $display calls and seems to be working fine. Will you be able to make a pull request ? I will cleanup the VCD reader code and hopefully that can be made available on Steve's website. It would be a nice addition to Icarus.

klammerj, 2017-01-01:
> Will you be able to make a pull request ? 

No need to. It's just a single patch. 

> I will cleanup the VCD reader code and hopefully that can 
> be made available on Steve's website. It would be a nice addition to Icarus.

+1
That would be great.

markNZed, 2017-01-01:
OK, I'm not sure how your change will get into the dev branch. Is that something you can look after ?

klammerj, 2017-01-01:
Usually I just wait for a while. Either it gets applied or not.

steveicarus, 2017-01-01:
Martin or I can apply it. If the patch is made with git tools, then you'll
get proper attribution.

klammerj, 2017-01-02:
Go ahead then, and apply it.

steveicarus, 2017-01-02:
Patch from klammerj applied to git master.

caryr, 2017-06-20:
This fix is not complete and is currently causing failures in two of the VVP tests (start_of_simtime1 and ro_synch) when compiled using clang or when using valgrind. The issue is that cb_data.obj is not set for these type of callbacks. The dynamic cast that was added can fail when it is given an undefined value.

steveicarus, 2017-06-20:
I encountered this problem in a real project (simbus) and fixed it there.
It would
certainly be better to initialize the .obj member in the vpi_member_cb call
whenever
the .obj is not required. I just couldn't tell when that was the case.

klammerj, 2017-06-25:
I've added some initialisations here, but it's all untested.
<https://github.com/klammerj/iverilog/commit/6b66b43376a44c72e00bd9e3f4b837be5840d25d.patch>

steveicarus, 2019-09-30:
What's the status of this report? It looks like it was fixed, then discovered to not be fixed after all, but it is unclear if there is a small example that demonstrates any remaining problem. So, can we decide if this is fixed or not? And if not, can we come up with a bite-sized example that demonstrates the problem?

caryr, 2019-10-01:
The proposed change has not been verified if it fixes the issue that is in the two tests I mentioned above.

steveicarus, 2019-10-05:
The proposed second patch is gone, and so far it is not clear to me that any real bug remains. I'm going to close this issue. I note that the original issue was fixed, and there only remain some question whether secondary bugs remain. If they prove to be real, please submit a new big report.

closed 2019-10-05 by steveicarus