steveicarus/iverilog #143: "assert" causes "malformed statement"

peterb12, 2017-01-10:
Using Icarus Verilog version 10.1 (stable) (v10_1_1)

With this code:
`module assert_tb;
   reg someValue;

   initial begin
      someValue = 0;
      assert(someValue == 0) else $error("Tragic.");
   end
endmodule // assert_tb`

I get this result:
`% iverilog -g2012 assert_tb.sv
assert_tb.sv:6: syntax error
assert_tb.sv:6: error: malformed statement`

peterb12, 2017-01-10:
I also pulled head-of-line trunk from today, same issue.

steveicarus, 2017-01-10:
Procedural assertions are not supported yet.

steveicarus, 2017-01-10:
Pushed a "sorry" message into the git master branch. It may be possible
to port that to the v10 branch. I'll try that later.

steveicarus, 2017-01-11:
I've ported this patch to the v10 branch (and pushed to git) so you should no get a reasonable "sorry" message for this. Actually adding support for the assert is a somewhat larger job.

closed 2017-01-11 by steveicarus