steveicarus/iverilog #144: Nested macro expansion not evaluating correctly

kdp1965, 2017-01-16:
I'm trying to get a design from one of the commercial simulators / synthesis tools to run with iverilog and running into a macro expansion issue with ivlpp (version 10.1).  The verilog code uses nested macro expansion which expands a single macro out to nearly 1000 signals.  I have included a much smaller snippet here:

```verilog
`define IO_HLPR_INPUT_2_VEC (name,c) \
   input [c-1:0] name``_0, \
   input [c-1:0] name``_1,

`define IO_HLPR_OUTPUT_2_VEC (n,c) \
   output [c-1:0] n``_0, \
   output [c-1:0] n``_1,

`define ADC_IO_SIGNALS(prefix)\
   `IO_HLPR_INPUT_2_VEC(prefix``bits,6) \
   `IO_HLPR_OUTPUT_2_VEC(prefix``data,10) \
   input prefix``clk

module adc_dsp
(
   `ADC_IO_SIGNALS(adc1_)
);
endmodule
```

This *should* evaluate to the following:

```verilog
module adc_dsp
(
   input [6-1:0] adc1_bits_0,
   input [6-1:0] adc1_bits_1,
   output [10-1:0] adc1_data_0,
   output [10-1:0] adc1_data_1,
   input adc1_clk
);
endmodule
```

However when I issue the command ivlpp macro.v (my test filename), the generated output after the macro expansion is:

```verilog
module adc_dsp
(
   
   (name,c)
   input [c-1:0] name_0,
   input [c-1:0] name_1,(adc1_bits,6)
   (n,c)
   output [c-1:0] n_0,
   output [c-1:0] n_1,(adc1_data,10)
   input adc1_clk
);
endmodule
```

Any assistance resolving this would be a great help and appreciated.  
Thanks,
Ken

kdp1965, 2017-01-16:
Hi Stephen,

Hey, I found the issue with the macro expansion after looking at the lex grammar and playing around with my macro.v file.  During macro definition (`define lines), it didn't like the whitespace between the macro name and the opening '(' token.  If I remove those spaces before the opening parenthesis, then it expands properly.

I fixed my local copy of the lex grammar by changing the lexor.lex file, line 346 to allow whitespace after the macro name.  I changed:

```
--- a/lexor.lex        2017-01-15 21:50:32.907034633 -0800
+++ b/lexor.lex     2017-01-15 21:46:16.603218707 -0800
@@ -343,7 +343,7 @@
             "cannot be directive keywords\n");
 }
 
-<DEF_NAME>[a-zA-Z_][a-zA-Z0-9_$]*"("{W}? { BEGIN(DEF_ARG); def_start(); }
+<DEF_NAME>[a-zA-Z_][a-zA-Z0-9_$]*{W}*"("{W}? { BEGIN(DEF_ARG); def_start(); }
 <DEF_NAME>[a-zA-Z_][a-zA-Z0-9_$]*{W}?    { BEGIN(DEF_TXT); def_start(); }
 
   /* define arg: <name> = <text> */
```

Then I had to trim the whitespace from the macro name during the macro definition in the 'define_macro' function, line 872:

```
@@ -869,6 +869,11 @@
     def->right = 0;
     def->up = 0;
     def->defaults = calloc(argc, sizeof(char*));
+    // Trim trailing space from name
+    idx = strlen(def->name);
+    while (def->name[idx-1] == ' ') {
+       def->name[--idx] = 0;
+    }
     for (idx = 0 ; idx < argc ; idx += 1) {
          if (def_argd[idx] == 0) {
                def->defaults[idx] = 0;
```

I don't know if this is the proper or most elegant way to fix this, but it seems to work at least.

Ken

steveicarus, 2017-01-16:
This looks entirely plausible. If  you make this as a git patch, I can
apply it to the git repository for you.

steveicarus, 2017-01-20:
I tested this against the regression test suite, and see that this
patch breaks this code:

`define USEC (`NSEC*1000)
`define MSEC (`USEC*1000)

So I did some digging, and in the IEEE1364-2005 standard, section 13.3.1,
where the `define directive is defined, there is this paragraph:

"If formal arguments are used, this list of formal argument names
shall be enclosed in parentheses following the make of the macro.
The formal argument names shall be simple_identifiers, separated
by commas and optionally whitespace. The left parenthesis shall
follow the text macro name immediately, with no space in between."

That is pretty cut-n-dry. You may be used to a more permissive
compiler, but since changing the compiler to go beyond the standard
in this manner will break existing code, I cannot these changes.

steveicarus, 2017-01-20:
The white space between the macro name and parameter lists in your `define directives is not correct according to the IEEE1364-2005 standard, and the IEEE1800 (SystemVerilog) inherits these rules. You may be used to a more permissive compiler, but for portability sake, I strongly recommend you adjust your directives.

Thanks for the report.

closed 2017-01-20 by steveicarus