steveicarus/iverilog #146: Constant function call in generate loop causes assert

dieterss, 2017-02-03:
If a generate loop has a call to a constant function inside, e.g.:
for (i = 0; i < log2(BUFS + 1); i = i + 1) ...
The compiler aborts with (latest git):
ivl: elab_scope.cc:1174: bool PGenerate::generate_scope_loop_(Design*, NetScope*): Assertion `test' failed.
Aborted

If the result of the constant function is first assigned to a parameter, e.g.:
parameter BUF2 = log2(BUFS + 1);
for (i = 0; i < BUF2; i = i + 1) ...
the compiler operates as expected and no assertion is hit.

A full testcase representing the issue is as follows:
--cut--
module test;

parameter [16 - 1:0] BUFS = 16'd1;

function integer log2;
input integer x;

integer i;
begin 
        i = 0; 
        while ((1 << i) < x)
                i = i + 1;
        log2 = i;
end
endfunction

wire [16 - 1:0] buf_seqno;
wire [log2(BUFS + 1) - 1:0] buf_seqno_gray;

genvar j;

generate 
        for (j = 0; j < log2(BUFS + 1); j = j + 1)
                assign buf_seqno_gray[j] = ^buf_seqno[log2(BUFS[(0 * 16)+:16] + 1) - 1:j];
endgenerate

endmodule
--cut--

klammerj, 2017-02-11:
They were missing a flag

Try this patch:
[0001-fix-for-const-eval.patch.txt](https://github.com/steveicarus/iverilog/files/768842/0001-fix-for-const-eval.patch.txt)
(hope it applies)

dieterss, 2017-02-16:
Hi klammerj,

Sorry for not getting back to you sooner. Your patch applied as follows:
$ patch -Np1 -i 0001-fix-for-const-eval.patch.txt 
patching file elab_scope.cc
Hunk #1 succeeded at 1169 (offset 14 lines).

The result works as I expect it to with my code.

Thanks.

klammerj, 2017-02-16:
Williams, can you apply the patch?

steveicarus, 2017-02-17:
Which branch? Looks like it will apply to the master devel branch
(and looks like it does the right thing) but don't know if that is what
you intend.

klammerj, 2017-02-17:
Yes, master is fine.

steveicarus, 2017-02-17:
Pushed to git master.

closed 2017-02-17 by steveicarus