steveicarus/iverilog #150: Handle attributes at module instantiation site

jersey99, 2017-03-16:
(* my_fancy_attribute *)
foobar1 foobar (clk(clk), rst(rst) ...);

Currently these attributes are recognized by the lexer but are ignored by the parser. I made the necessary modifications so that these attributes are "handled". [here](https://github.com/steveicarus/iverilog/pull/149). I would love to get some feedback, and see this feature through completion.

I believe this is an important feature to have as this enhances automatic code generation. Eg: Automatically setting registers sitting inside nested modules from a local addr/data bus that is available at the top_level module.

steveicarus, 2017-03-16:
Didn't I just merge a pull request that does this for you?

jersey99, 2017-03-16:
You did. Thanks!

I opened the issue and PR nearly simultaneously. I intended this location to be a place to hold the discussion.

I am closing the issue.

closed 2017-03-16 by jersey99