steveicarus/iverilog #153: Running VPI on Windows

e19293001, 2017-04-06:
I've been trying to run vpi in windows by following the
instructions discussed in
http://iverilog.wikia.com/wiki/Using_VPI.

I get this error when compiling:

    gcc -c -fpic hello.c -I/c/iverilog/include/iverilog
    hello.c:1:0: warning: -fpic ignored for target (all code is position independent)
     # include  <vpi_user.h>
     ^
    gcc -shared -o hello.vpi hello.o -lvpi -L/c/iverilog/lib
    hello.o:hello.c:(.text+0x18): undefined reference to `vpi_printf'
    hello.o:hello.c:(.text+0x5a): undefined reference to `vpi_register_systf'
    collect2.exe: error: ld returned 1 exit status
    make: *** [vpi] Error 1

I know that I have to do this with DLLs but could someone share
the correct way to compile? 

Thanks

matthamptonasic, 2017-04-06:
The -I passed to gcc should indicate where the vpi_user.h can be found.  Is /c/iverilog/include/iverilog the correct location on your system?  I believe it's in the top level of your Icarus install path.

I have a 'make' setup on my system, but an example of the compile command to compile my vpi base class is:
`g++ -fPIC -std=c++11 -c -I./ -I../BitVector -I../Common -I../DataTypes -I../Environment -I../Event -I../Logging -I../Test -I/usr/local/bin/iverilog_v10 -o /home/matt/dump/Hardware/_common/fifo/cpp/vpi.o vpi.cc`
The last -I has the location of my vpi_user.h.

It may not help much, but I have a (relatively small) working system here:
[https://github.com/matthamptonasic/Hardware](url)

steveicarus, 2017-04-06:
The iverilog_vpi command should take care of all the command line
muckery needed to compile a vpi module. Does it not work on windows?

e19293001, 2017-04-06:
When I use iverilog-vpi, I get the following:

    $ iverilog-vpi hello.c
    hello.c: In function 'hello_compiletf':
    hello.c:3:33: warning: unused parameter 'user_data' [-Wunused-parameter]
     static int hello_compiletf(char*user_data)
                                     ^
    hello.c: In function 'hello_calltf':
    hello.c:8:30: warning: unused parameter 'user_data' [-Wunused-parameter]
     static int hello_calltf(char*user_data)
                                  ^
    hello.c: At top level:
    hello.c:14:6: warning: function declaration isn't a prototype [-Wstrict-prototypes]
     void hello_register()
          ^
    hello.c:27:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
     void (*vlog_startup_routines[])() = {
     ^
    hello.o: In function `hello_register':
    c:/example_vpi_windows/hello.c:24: undefined reference to `vpi_register_systf'
    hello.o: In function `hello_calltf':
    c:/example_vpi_windows/hello.c:10: undefined reference to `vpi_printf'
    collect2.exe: error: ld returned 1 exit status
    Compiling hello.c...
    Making hello.vpi from  hello.o...

Does iverilog-vpi generates DLLs to have a shared library?

martinwhitaker, 2017-04-06:
e19293001 wrote:
> When I use iverilog-vpi, I get the following:
>
>     $ iverilog-vpi hello.c
>     hello.c: In function 'hello_compiletf':
>     hello.c:3:33: warning: unused parameter 'user_data' [...]
The .vpi file is a DLL that gets loaded when you run vvp.

What version of Windows are you using?
What version of Icarus Verilog are you using?
Did you install from one of Pablo's binary packages or build from source?
Are you using MSYS2 or Cygwin? (I'm assuming from your shell prompt that 
you are using one of these).

I last tested this a while ago, but following the instructions at

   http://iverilog.wikia.com/wiki/Installation_using_MSYS2

enabled me to run the VPI tests in the test suite (checkout 
https://github.com/steveicarus/ivtest and run 'perl vpi_reg.pl' in the 
ivtest directory).

Martin

e19293001, 2017-04-06:
I'm using Windows 10 and iverilog-10.1.1 which is located in this
link:

http://bleyer.org/icarus/iverilog-10.1.1-x64_setup.exe

I've install this Pablo's binary package and using MSYS32.

I'm going to try to install from source using the link you've
gave.

Thank you.

e19293001, 2017-04-06:
Problem fixed by compiling from source obtained from
ftp://ftp.icarus.com/pub/eda/verilog/v10/

I suspect that the problem occured on environment
incompatibilty. Since I used the 64-bit version of iverilog
obtained Pablo's binary package and used a 32-bit version of
MSYS. Though I'm not sure about this. The problem has been solved
anyway. Thank you!

closed 2017-04-06 by e19293001