steveicarus/iverilog #155: Numbers in ROM file not separated by blanks are not considered several ones

piranna, 2017-04-15:
On a ROM file read by `readmemh()` function, if you don't separate hexadecimal values with spaces they are considered a single number. For example, `00000000` is considered a single byte with value zero instead of 4 bytes.

martinwhitaker, 2017-04-21:
That is correct. The IEEE standard states "White space and/or comments shall be used to separate the numbers." (1364-2005 section 17.2.9). What the standard doesn't say is what should happen if you supply a number that is too large to fit in the memory word. Icarus handles this in the same way as it would handle an assignment (i.e. discarding the excess bits).

piranna, 2017-04-22:
Shouldn't it throw a warning instead of just silently discard them in both cases?

steveicarus, 2019-09-30:
I've added a warning in git master.

closed 2019-09-30 by steveicarus