steveicarus/iverilog #156: Dump parameter and localparam in vcd

robsmi1970nyc, 2017-05-07:
Incisive produces:

```
$scope module mymodule $end
$var parameter 32 !    BITS  $end
...

$dumpvars
b10 !
...

[no further changes]
```

for both `parameter BITS = 2` and `localparam BITS = 2` and this seems to be specified by the Verilog 2005 standard accordingly.

martinwhitaker, 2017-05-07:
The standard is somewhat inconsistent. In section 18 it states "A VCD file contains information about value changes on selected variables in the design" and in section 4.10 it states "Parameters are not variables; they are constants", so it would be reasonable to conclude that parameters are not dumped. But as you highlight, the syntax for $var includes "parameter" as a possible var_type.

Also, the $dumpvars section is only supposed to contain variables that change value in a given time step, and parameters never change value at run time.

jotego, 2021-05-03:
Is there an alternative way to dump parameter values?

PR #714 (2022-05-17): Add parameters to vcd dumps
Writing parameters into VCD files makes the values available to waveform
tools. This can be done easily enough by writing out a $dumpadd section
at the beginning of the file that sets the parameter values. We don't need
to track the values over change, because by definition they do not change.

This changes the typical vcd output as well, so a few of the regression tests
need to be adjusted to account for this.

Also, while tracking this down, found and fixed the vvp/README.txt documention
for the .param/x records.

This fixes #156

TODO: Fix the fst dumper as well.
closed 2022-05-17 by steveicarus