steveicarus/iverilog #157: Give an error if a localparam is modified at instantiation

robsmi1970nyc, 2017-05-07:
iverilog seems to treat localparam just like parameter, which makes its purpose of having a private parameter useless.

Incisive complains and exists an error as expected.

module.v:

```
module mymodule;
localparam ABC = 2
...
```

module_tb.v:

```
module mymodule_tb;
mymodule_tb #( .ABC(123) ) top;
...
```

marshallversteeg, 2017-05-15:
I reproduced your sample on VCS and found that VCS only gives a warning, not an error, and doesn't override the localparam. 

The correct behavior is defined as "Local parameters are identical to parameters except that they cannot directly be modified by ... instance parameter value assignments." in IEEE 1800-2009.

So VCS and Incisive more closely match the correct behavior whereas IVerilog is doing something incorrectly.

martinwhitaker, 2017-06-16:
Fixed in both the master and v10 branches. I've modified the existing check for whether an overridden parameter exists to only look at parameters. This is not as helpful as it could be - it warns you the parameter doesn't exist rather than warning you that you are overriding a localparam - but it would need a more extensive change to the compiler to do better.

closed 2017-06-16 by martinwhitaker