steveicarus/iverilog #158: cannot schdule named event with vpi_put_value

makechip, 2017-06-05:
Attempting to schedule a named event with vpi_put_value() gives this runtime error:
ERROR: vpip_time_units_from_handle called with object handle type=34
vvp: vpi_priv.cc:489: int vpip_time_units_from_handle(vpiHandle): Assertion `0' failed.

The vpip_time_units_from_handle() method doesn't have a case for a named event.

klammerj, 2017-06-05:
does this help?

[0001-NamedEvent-fix.txt](https://github.com/steveicarus/iverilog/files/1052781/0001-NamedEvent-fix.txt)

makechip, 2017-06-06:
The patch compiles okay, but now I get a seg-faulton the first attemt to schedule an event:Program received signal SIGSEGV, Segmentation fault.vpi_put_value (obj=0xa1c160, vp=0x0, when=0x7ffff5f4df20 <edgeTimes+15360>, flags=4) at vpi_priv.cc:11111111                put->value = *vp;(gdb) backtrace#0  vpi_put_value (obj=0xa1c160, vp=0x0, when=0x7ffff5f4df20 <edgeTimes+15360>, flags=4) at vpi_priv.cc:1111#1  0x00007ffff5c9fd1c in schedulex (channel=channel@entry=20, index=index@entry=3536, numEdges=<optimized out>,     edgeTime=0xa1ae80, oen_h=oen_h@entry=0xa12090, drv_h=drv_h@entry=0xa11fb8, stb_h=0xa1c160, exp_h=0xa12048)    at pecard_timing.c:4569#2  0x00007ffff5ca01d4 in pecard_timing_calltf (user_data=<optimized out>) at pecard_timing.c:4684#3  0x00000000004b8e26 in vpip_execute_vpi_call (thr=thr@entry=0xa446c0, ref=0xa44640) at vpi_tasks.cc:951#4  0x0000000000479f71 in of_VPI_CALL (thr=0xa446c0, cp=<optimized out>) at vthread.cc:6020#5  0x00000000004722e2 in vthread_run (thr=0xa446c0) at vthread.cc:706#6  0x0000000000485d84 in schedule_simulate () at schedule.cc:1183#7  0x0000000000444359 in main (argc=6, argv=0x7fffffffe068) at main.cc:501I Think the problem is that for an event, the vpi_value is set to NULL, butat line 1111 of vpi_priv.cc, it tries to read the value of the NULL pointer.-Eric

vpi_put_value (#158)

klammerj, 2017-06-09:
Can you post a minimal testcase? 
I'll have to dig around in schedulex etc. to figure out where the vp=0x0 is coming from.

makechip, 2017-06-09:
I will work on a minimal test case, and send it sometime this weekend.I believe the vp=0x0 is coming from me, because I am setting the vpi_valueto NULL when scheduling a namedEvent.  I tried setting the value to vpi1,and it didn't crash, but I didn't get any events either.-Eric

martinwhitaker, 2017-06-15:
This should now be fixed in both the master and v10 branches. Please test that it works for your code.

makechip, 2017-06-15:
Thanks.  I will try it out in a day or two, and report back.-Eric

vpi_put_value (#158)

makechip, 2017-06-16:
Great! Thank you very much. "It works for me."-Eric

martinwhitaker, 2017-06-16:
Thanks for confirming the fix. Closing.

closed 2017-06-16 by martinwhitaker