steveicarus/iverilog #159: always_comb, always_ff, always_latch

AlexDaniel, 2017-07-05:
There has to be an issue for this already, but for some reason I am unable to find it. I see that it was mentioned in issue https://github.com/steveicarus/iverilog/issues/97, but I see no follow-up to this.

I think out of all three the main priority is for ``always_comb`` and ``always_ff``, I don't think ``always_latch`` is used very often (my best guess is that these two are easier to implement also).

caryr, 2017-11-20:
Preliminary support for these has been added to the development branch. This should be fully functional except always_comb and always_latch do not currently have support for the special time zero trigger and none of the special compile time checks are currently implemented. I will be working on the time zero trigger next to get correct functionality.

caryr, 2017-12-04:
I have added support for a time zero trigger. It triggers in the first inactive region. All the various checks are still in progress.

caryr, 2017-12-17:
Over the past few weeks I have made extensive changes and code additions to warn about constructs that cannot be synthesized in the always_* processes. I have a few more issues marked with questions that I would like to get finished before I push this next part of the implementation. After this I still need to add the variable checking and the checks to see if the statements match the type of logic implied by the process. I also need to hide the T0 trigger in VVP. Some of the questions are subtle and may take time to answer.

caryr, 2017-12-27:
I have pushed some more changes. Hiding the T0 trigger, checking that L-values are assigned correctly, and that the structure is correct has not been started. There are also a few thing that need to be fixed in the basic synthesis checks tasks/function are not looked at and there are a few other questions that I have regarding other items (e.g. what exactly is allowed in a synthesized for statement).

This needed to get pushed because it is a fairly extensive change and I wanted to get it available for others to start testing.

Hopefully more changes coming soon.

caryr, 2018-09-24:
I pushed some changes that I started some time ago. Now that this is finished my next task, when I have some free time, is to fix the sensitivity list calculation and then start working on the check to verify variables are only set from the one always block.

My work schedule for the last few years has allowed very little free time to work on Icarus and that will likely not change going forward so my progress will be slow.

red0bear, 2019-02-04:
@zyedidia Last time i tested ... it was working fine.

martinwhitaker, 2019-02-04:
@zyedidia, have you enabled SystemVerilog support using the -g option (e.g. -g2012) ?

martinwhitaker, 2019-02-04:
That example compiles without error here, using exactly the same iverilog version. What version of bison are you using? Check the parse.output file in the top-level directory for any warnings.

zyedidia, 2019-02-04:
I'm using bison version 2.3 (the default bison that was installed on my Macbook).

martinwhitaker, 2019-02-05:
OK, that's a fairly ancient version. Not that I have any reason to believe it shouldn't work, but it seems the most probable cause of the different behaviour. Can you attach the parse.output file it generated,

zyedidia, 2019-02-06:
I uploaded my parse.output file. I just compiled with autoconf, configure and make and used the binary in the driver folder. This isn't a pressing issue for me but it's just a bit strange that my build doesn't work.

[parse.output.zip](https://github.com/steveicarus/iverilog/files/2837712/parse.output.zip)

kkiningh, 2019-06-07:
I had the same problem - turns out I had an old version of icarus-verilog installed through brew that was getting picked up by the iverilog driver. Removing it fixed the problem.

caryr, 2019-09-30:
I have pushed more support for always_*. Constant array selects should work correctly, though I am still working to get packed array selects working correctly. A sorry message is printed for this case so this unsupported functionality will be reported to the user. Constant packed selects along with making sure a value is only written from a single process still needs to be added. If other issues are found please report them.

mciepluc, 2019-12-16:
Is this expected to work in 10.3 release?

martinwhitaker, 2019-12-19:
@mciepluc, no. New features are not normally backported to the stable release.

bman12three4, 2022-05-19:
Has there been any progress on getting packed array/struct selects working?

jmerdich, 2022-05-29:
Just leaving a hack here and similarly expressing interest in full support for `always_comb` and friends. 

```verilog
`ifdef __ICARUS__
// Waiting on https://github.com/steveicarus/iverilog/issues/159 for array select support :(
`define ALWAYS_COMB_ICARUS_159 always @(*)
`define ALWAYS_FF_ICARUS_159 always @(*)
`else
`define ALWAYS_COMB_ICARUS_159 always_comb
`define ALWAYS_FF_ICARUS_159 always_ff
`endif
```

steveicarus, 2022-12-16:
This issue, so far as it is reported, seems to not exist in the current development. Sop closing this issue.

closed 2022-12-16 by steveicarus