steveicarus/iverilog #161: assertion fails with struct inside struct

kkamundsen, 2017-08-09:
Hi,

I  am trying to use some verilog code that uses system verilog structs inside structs. I have boiled my code down to bare minimum, but I still cannot get it to compile in iverilog. The code compiles and runs in ModelSim. I have tried the newest master branch on github of iverilog.

In the attached file I have commented out the four failing lines. By removing the comments you can see these errors when compiling the module:
test.v:18: assert: elab_lval.cc:1169: failed assertion name_idx->name == member_name
test.v:19: assert: elab_lval.cc:1169: failed assertion name_idx->name == member_name
test.v:21: error: Unable to bind wire/reg/memory `main.sub.sub_local' in `test'
test.v:22: error: Unable to bind wire/reg/memory `main.sub_list['sd0].sub_local' in `test'

To compile the test, I have been using this command line
iverilog -g2012 test.v

The file test.v is found here:
[test.v.txt](https://github.com/steveicarus/iverilog/files/1211467/test.v.txt)

I hope you can help.

Best regards,

Kai Kristian

martinwhitaker, 2017-09-08:
Sorry for the slow response. I did take a look at this, but at present the compiler really doesn't support nested structs, and it's going to take a fair amount of work to make it do so.

steveicarus, 2019-09-24:
I've added variations of the sample test into the ivtest suite (gh161a and gh161b) and I've pushed to git master a fix to add this feature.

closed 2019-09-24 by steveicarus