steveicarus/iverilog #164: Assertion failure while copying dynamic array

jleahy, 2017-10-02:
The following code:

```
module test;

reg[7:0] mema[];
reg[7:0] memb[];

initial begin
  mema = new[4];
  memb = new[4] (mema);
end

endmodule
```

Results in the following error when compiling with "-g2012".

```
test.sv:8: internal_error: net type doesn't match context type.
test.sv:8:               : net type=dynamic array of logic unsigned[7:0]
test.sv:8:               : context type=logic unsigned[7:0]
test.sv:8: assert: elab_expr.cc:3475: failed assertion ntype->type_compatible(net->net_type())
```

martinwhitaker, 2017-10-07:
It appears that Icarus expects the initial value to be of an elementary type, which will be used to initialise every element of the array. I don't know why Steve did this, as from my reading of the IEEE standard, the initial value should be a dynamic array.

martinwhitaker, 2017-10-08:
I've added support for initialising from another dynamic array in both the master and v10 branches. I've left the old Icarus behaviour intact.

closed 2017-10-08 by martinwhitaker